XCSoar/XCSoar #722: Airspace Luxembourg (CTR) Problem

fraktal-sb, 2021-09-19:
XCSoar versions having and not having the problem
-------------------------------------------------

OK: 6.8.17
NOK: 7.19

Expected behavior
-----------------

Airspace should be accepted

Actual behavior
---------------

When using the an actual airspace file with the Luxembourg Airspace XCSoar issues an error message but continues the first time. When XCSoar ist closed and reopened again no message is issued and XCSoar goes back to the splash screen. The only way to get XCSoar running again is to remove the airpace File from the directory "Airspace" and and then restart XCSoar.

[image]

Steps to reproduce the behavior
-------------------------------

use  https://www.lvzc.be/index.php/luchtruim/luchtruim-4 (contains Luxembourg CTR Airspace) or 

or the following test file which contains only Luxembourg Airspace

[CTR_TEST.txt](https://github.com/XCSoar/XCSoar/files/7191711/CTR_TEST.txt)

Do you have any idea what may have caused this?
-----------------------------------------------

no

Do you have an idea how to solve the issue?
-------------------------------------------

no

August2111, 2021-09-20:
In the OpenAir text format a central point of a circle or an arc is defined beyond the limits of an airspace definition - up to a next declaration... Inside the new code there is a check of a valid centre point, but it shouldn't be not valid if such a point is defined in a former airspace. This is really strange because you cannot change the order of airspaces! Normally most of the airspace define all this points inside the declaration - but in the definition of drawing style in the beginning of the file - from my point of view used in the old WinPilot only! - there is only one declaration of central point in the 1st airspace definition, but inside every drawing definition is defined a circle ('DC 0.05')  ;-(
To follow the (old) rules there is to declare the central point inside an airspace from a 'global' central point ...
@fraktal-sb : From my point of view a good way to get a running XCSoar is to remove the '***Winpilot COLOR-Table***' part from the airspace file (f.e line 250 to 302 in BELLUX_WEEKEND_2021-03-21b.txt) - because in most of the cases/application it is never used anymore!

MaxKellermann, 2021-09-20:
The reason for this error message is a broken airspace file. The airspace on line 262 is missing the "center" location (`V X=...`), as the error message clearly states - and indeed!
```
AC Q
SP 0,2,255,128,0
SB -1,-1,-1
DC 0.05
```
Version 6.8.17 is not "OK", it just ignored this error silently and continues with broken data (center assumed at latitude 0, longitude 0, which is a mistake, a bug in the old XCSoar version).

closed 2021-09-20 by MaxKellermann