XCSoar/XCSoar #723: Docker build not working (Target Android)

groundhogxc, 2021-09-21:
The docker build for target Android fails, with (at least) two problems:

What I try to do:
* Run docker interactively
* Command "make TARGET=ANDROID"

2 Problems:

1) The first error is a missing usbserial library. This can be fixed with:
git submodule init
git submodule update

2) The second error is 
make: *** no rule to make target 'lib/glm/glm/detail/type_int.hpp', which is needed by 'output/ANDROID/armeabi-v7a/dbg/src/Renderer/GlassRenderer.o'. Stop.

This one I do not understand. The glm subdirs are there, but the file "type_int.hpp" does not exist; but I also do not see why it would be needed in "GlassRenderer", it is not inculded there (or anywhere else as far as I can see). Must be something in the makefile logic, but I do not aunderstand it.

Versions: 7.18, as well as current pull

MaxKellermann, 2021-09-21:
The first error is your error, not XCSoar's.
The second error is outdated dependency files; you previously built with a different GLM version, and the dependency files automatically created during that build refer to files from that GLM version; the new version has probably renamed some internal files, and that one file is gone now. You can fix that by doing a clean build.

groundhogxc, 2021-10-06:
Oops, my mistake.
(With the first one though I had thought the provisioning system is supposed prepare the entire build process, so that can then be used by the automatied pipelines. Apparently not.)
Sorry for the hassle...