XCSoar/XCSoar #724: Time zone not working no xcsoar 7.18 for kobo.

marck302, 2021-09-21:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

GrantHeaney, 2021-09-22:
Kobo Glo HD, just tested it, seems to be working for me. 
Menu   configuration/system/Time
Use GPS time set to on, can change UTC offset to anything i like.

MaxKellermann, 2021-09-22:
This bug report is empty.

closed 2021-09-22 by MaxKellermann