XCSoar/XCSoar #725: Error handling Airspace files

fraktal-sb, 2021-09-22:
XCSoar versions having and not having the problem
-------------------------------------------------

OK 6.8.17
NOK 7.19

Expected behavior
-----------------

XCSoar should allwas issue an error message on corrupted data.

Actual behavior
---------------

XCSoar issues at the first contact with corrupted airspace data an error message and then continues operation. Next time XCSoar is restarted no more error messages are issued and the Program goes back to the splash screen. This is confusing and inconsistent.

Steps to reproduce the behavior
-------------------------------

use for example one of the (corrupted) airspace files in https://github.com/XCSoar/XCSoar/issues/722. Open XCSoar again and go to "SIM". You will be redirected to the splash screen. Use UNIX or KOBO version

Do you have any idea what may have caused this?
-----------------------------------------------

bug in airspace error handling

Do you have an idea how to solve the issue?
-------------------------------------------

correct the bug

MaxKellermann, 2021-09-24:
I can't reproduce your problem. I get an error message every time I start XCSoar with this corrupt airspace file.
Please bisect the problem to the faulty commit. I can't help until I have the exact first commit.

fraktal-sb, 2021-09-25:
Problem must be related to presence of a map file in site configuration. Without a map file I allways get the error message. With a map file the behavior is as above decribed. Tested on UNIX and KOBO version, see screenshots.

[image]
[image]
[image]

fraktal-sb, 2021-09-28:
first occurence of the problem is with version 7.18
also testet with versions 7.5, 7.11, 7.14, 7.16, 7.17 which are OK

closed 2021-10-04 by MaxKellermann