XCSoar/XCSoar #726: Time zone not working on xcsoar 7.18 for kobo

marck302, 2021-09-23:
XCSoar versions having and not having the problem
-------------------------------------------------

7.18
Which XCSoar version is the last one you know that did not show the problem?
-->7.16

Expected behavior
-----------------

 should adjust time to selected time zone

Actual behavior
---------------

 the selecteted time zone return to "zero" when turn off and turn on again

Steps to reproduce the behavior
-------------------------------

just adjust time zone and restart the kobo, mostly happens in new installations

Do you have any idea what may have caused this?
-----------------------------------------------

 new version

Do you have an idea how to solve the issue?
-------------------------------------------

no

MaxKellermann, 2021-09-23:
I have no idea what this bug report is about. Add an accurate and exact description of the issue.
And you forgot to test version 7.17.

marck302, 2021-09-23:
Version 7.17 don’t even worked on kobo it crashed almos on startup

Enviado do Email<https://go.microsoft.com/fwlink/?LinkId=550986> para Windows 10

marck302, 2021-09-23:
It, about time zone, the  local hour!!!!

Enviado do Email<https://go.microsoft.com/fwlink/?LinkId=550986> para Windows 10

MaxKellermann, 2021-09-23:
Sigh, still no accurate description. Sorry, can't help you.

closed 2021-09-23 by MaxKellermann

marck302, 2021-09-23:
Ok brow, all the pilots here with same problem, but it’s not so importante, think someone will fix someday

Enviado do Email<https://go.microsoft.com/fwlink/?LinkId=550986> para Windows 10