XCSoar/XCSoar #727: XCSoar v7.16 and Android 10, display slowness

dbsonic1, 2021-09-23:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

MaxKellermann, 2021-09-23:
You're using an old version. Update to the latest before reporting a bug.

dbsonic1, 2021-09-23:
Okay, I'll update to v7.18 and report back.

MaxKellermann, 2021-09-23:
And try to reproduce it with replay or simulator.

dbsonic1, 2021-09-24:
None of the replay attempts below seemed to recreate the issue.  It seems the only way would be to take a flight and capture a log to give to you.

v7.18 replay of flight upon which issue was observed
v7.16 replay of flight upon which issue was observed
v7.16 replay of flight upon which issue was observed and heated device to 120+ F
v7.16 replay of flight upon which issue was observed and cooled device to 40ish F
v7.16 replay of flight upon which issue was observed in cockpit with power supply applied. 
v7.16 fly mode in cockpit but not moving(sitting on ground)

lordfolken, 2021-12-09:
I will close this issue for now. If you find a way of how to reproduce the problem, then please reopen.

closed 2021-12-09 by lordfolken