XCSoar/XCSoar #729: USB (OTG) and an FTDI serial  bridge crashes XCS on connect on Android

hjr, 2021-09-23:
XCSoar versions having and not having the problem
-------------------------------------------------
Tried 7.18, 7.17, and 7.12 all with the same behavior. On an Android  9 device

Expected behavior
-----------------
Plugging the USB serial dongle the device comments with "usb connected ...". Starting XCSoar -> config -> NMEA. Edit the first port and select the listed "USB to UART bridge". Hit select (Flarm shows up as driver). The Android then asks: "Allow XCS to access this USB toy...", hit "Yes".
And all should be fine.

Actual behavior
---------------
but XCS crashes in this moment. Onces configured like this the restart in fly mode directly crashes again. Restart into sim mode does work and let you undo the NMEA configuration. 

Steps to reproduce the behavior
-------------------------------
As described above, always reproducable. The Android is a Boox eReader Poke 2.

Do you have any idea what may have caused this?
-----------------------------------------------
Only the connected and configured USB dongle get XCS crashing. Unplugging the USB dongle is sensed and commented below the configuration with:" IO exception. Device not found. Which makes sense.

Do you have an idea how to solve the issue?
-------------------------------------------
no

MaxKellermann, 2021-09-23:
logcat?

hjr, 2021-09-24:
I fear not very helpful ..
[xcsoar.log](https://github.com/XCSoar/XCSoar/files/7223624/xcsoar.log)

hjr, 2021-09-24:
Here just the sequence: Start XCS, activate the malicious nmea source.
[xcsoar.log](https://github.com/XCSoar/XCSoar/files/7223923/xcsoar.log)

MaxKellermann, 2021-09-24:
That's a xcsoar log file (without any crash info), not a logcat. logcat is Android's internal log, which can be extracted using the Andorid Debug Bridge (adb) from a computer connected to the phone.
What would also be useful is a NMEA dump of this XCVario.

hjr, 2021-09-24:
Would like to send an logcat, have adb and the Android connected, but dont know how to connect the XCV over USB the same time. Any hint on that?

[2021-09-24_20-42.zip](https://github.com/XCSoar/XCSoar/files/7228442/2021-09-24_20-42.zip)

MaxKellermann, 2021-09-24:
`adb logcat` will print past crashes as well. First let it crash, then connect to computer and extract the crash log.

hjr, 2021-09-25:
This is a short one:
adb logcat xcsoar:V *:S
- waiting for device -
--------- beginning of main
--------- beginning of system
--------- beginning of crash

hjr, 2021-09-25:
[xcsoar_crash_on_usb_connect_logcat.txt](https://github.com/XCSoar/XCSoar/files/7229296/xcsoar_crash_on_usb_connect_logcat.txt)

Yip, and here is some more info :)

MaxKellermann, 2021-09-25:
Yes, that's the right stuff!
```
FATAL EXCEPTION: Thread-2
Process: org.xcsoar, PID: 2489
java.lang.NullPointerException: Attempt to get length of null array
	at com.felhr.usbserial.Buffer.write(Buffer.java:13)
	at com.felhr.usbserial.SerialBuffer$SynchronizedBuffer.put(SerialBuffer.java:107)
	at com.felhr.usbserial.SerialBuffer.putWriteBuffer(SerialBuffer.java:77)
	at com.felhr.usbserial.UsbSerialDevice.write(UsbSerialDevice.java:133)
	at org.xcsoar.UsbSerialPort.write(UsbSerialPort.java:128)
```
I identified the bug, will be fixed soon.

closed 2021-09-26 by MaxKellermann