XCSoar/XCSoar #736: System crash on startup - Android with SensBox

lyusupov, 2021-10-01:
XCSoar versions having and not having the problem
-------------------------------------------------

Device: Android 5.0 ARM64
BLE data source: SensBox

7.12 - NO
7.18 - YES

Expected behavior
-----------------

Shouldn't crash at no way

Steps to reproduce the behavior
-------------------------------

Select Sensbox BLE ID in Android device settings as 'BLE sensor' device. 

Do you have any idea what may have caused this?
-----------------------------------------------

Excerpt from 'adb logcat':

```
E/JavaBinder(17617): *** Uncaught remote exception!  (Exceptions are not yet supported across processes.)
E/JavaBinder(17617): java.lang.NoSuchMethodError: No static method toUnsignedLong(I)J in class Ljava/lang/Integer; or its super classes (declaration of 'java.lang.Integer' appears in /system/framework/core-libart.jar)
E/JavaBinder(17617):    at org.xcsoar.BluetoothSensor.onCharacteristicChanged(BluetoothSensor.java:174)
E/JavaBinder(17617):    at android.bluetooth.BluetoothGatt$1.onNotify(BluetoothGatt.java:487)
E/JavaBinder(17617):    at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:399)
E/JavaBinder(17617):    at android.os.Binder.execTransact(Binder.java:451)
```

Do you have an idea how to solve the issue?
-------------------------------------------

No.

MaxKellermann, 2021-10-02:
The method is a Java 8 feature, but that Android version only has Java 7. We need to revert to a Java 7 API.

MaxKellermann, 2021-10-02:
https://docs.oracle.com/javase/8/docs/api/java/lang/Integer.html#toUnsignedLong-int-