XCSoar/XCSoar #738: Airspace reading error !

doramail, 2021-10-02:
Which XCSoar version are you using when seeing the problem? PC & Android Version 7.19
Which XCSoar version is the last one you know that did not show the problem? PC & Android version 7.18

Expected behavior
-----------------

Expected behavior should be no error message displayed about airspace files when program starts.

Actual behavior
---------------

After the update of the PC version & Android versions, I got this message displayed when launching the program.

[image]

Steps to reproduce the behavior
-------------------------------

Just load the program with the french airspace file.

Do you have any idea what may have caused this?
-----------------------------------------------

There should be an issue about the DC values but I do not know more. 

Do you have an idea how to solve the issue?
-------------------------------------------

Unfortunately no.

MaxKellermann, 2021-10-02:
This just means that your airspace file is broken. Previously such errors were ignored silently, which can cause problems because some airspaces may be missing; it's always better to fix the airspace files than ignoring the existence of those airspaces.

MaxKellermann, 2021-10-02:
Duplicate of https://github.com/XCSoar/XCSoar/issues/722

closed 2021-10-02 by MaxKellermann

doramail, 2021-10-02:
OK, thank you for your super fast answer, I am always impressed with your fast reaction to issues we may send you. 
I have corrected the original airspace so, now there are no more errors. 
Best regards.

doramail, 2021-11-03:
Super, OK, thank you for the info.
Best regards.