XCSoar/XCSoar #740: Incorrect behaviour when 'Score start exit' = OFF (Entry)

mostlyharmlessme, 2021-10-03:
XCSoar versions having and not having the problem
-------------------------------------------------
Feature was recently introduced at v7.17 by @lordfolken and issue seen in v7.19.

Expected behavior
-----------------
Paragliding competitions often have Start cylinders which are scored on ENTRY rather than the XCSoar default of EXIT.
v7.17 added the task rules option "Score start exit:ON/OFF", however these are a few problem scenarios when 'Score start exit' = OFF (Entry).

**Scenario 1:-**
- Start radius = 2km
- T1 cylinder radius = 0.5km located **outside** start cylinder
- Take-off = **Outside** start cylinder

Expected that 'Next distance' to start would count down to 0km as start cylinder crossed.

**Scenario 2:-**
- Start radius = 2km
- T1 cylinder radius = 0.5km located **inside** start cylinder
- Take-off = **Outside** start cylinder

Expected that after start was scored, 'Next distance' to T1 would count down to 0km when approaching T1 cylinder.

**Scenario 3:-**
- Start radius = 2km
- T1 cylinder radius = 0.5km located **outside** start cylinder
- Take-off = **Inside** start cylinder

Expected that start would not be scored when exiting the start cylinder but when it is re-entered after a 180deg turn, then 'Next distance' to T1 would count down to 0km when approaching T1 cylinder.

**Scenario 4:-**
- Start radius = 2km
- T1 cylinder radius = 0.5km located **inside** start cylinder
- Take-off = **Inside** start cylinder

Expected that start would not be scored when exiting the start cylinder but when re-entered after a 180deg turn, then 'Next distance' to T1 would count down to 0km when approaching T1 cylinder.

Actual behavior
---------------
**Scenario 1:-**
'Next distance' to start value randomly toggles between the correct value (x km) and a value increased by the diameter of the start cylinder (x+4 km).

**Scenario 2:-**
'Next distance' to T1 value randomly toggles between the correct value (x km) and a value increased by the diameter of T1 (x+1 km).

**Scenario 3:-**
Start is scored incorrectly when exiting the start cylinder.

**Scenario 4:-**
Start is scored incorrectly when exiting the start cylinder (like scenario 3) then 'Next distance' to T1 value randomly toggles between the correct value (x km) and a value increased by the diameter of T1 (x+1 km).

Steps to reproduce the behavior
-------------------------------

Set up tasks as given in the scenarios paying particular attention to whether take-off and turnpoint 1 (T1) are inside or outside the start cylinder. When T1 is inside the start cylinder I have generally used the same waypoint for both. I used the simulator.

Do you have any idea what may have caused this?
-----------------------------------------------
No.

Do you have an idea how to solve the issue?
-------------------------------------------
No.

lordfolken, 2021-10-04:
I have no knowledge of this, nor did i implement said feature.

mostlyharmlessme, 2021-10-04:
Sorry - must have picked up the wrong name. My apologies.

= OFF (Entry) (#740)

MaxKellermann, 2024-02-28:
I tried to reproduce your problem.
1.+2. I saw no "toggling", I always saw the correct distance to the cylinder boundary
3.+4. Start is not scored when exiting the start cylinder, only when re-entering it

I tried XCSoar 7.17, 7.19 and 7.42. All versions showed correct behavior.

If you insist that this problem does exist, post your task and an IGC file which demonstrates the bug, and we'll reopen the report.

closed 2024-02-28 by MaxKellermann

mostlyharmlessme, 2024-02-28:
Thanks for looking into this. It was a while ago and I no longer have 
the instruments to repeat the test.
I'm happy the issue is closed as a non-problem.

Cheers,
Mark

From "Max Kellermann" ***@***.***>

Date 28/02/2024 12:18:11