XCSoar/XCSoar #741: Kobo Version 7.19 compile errors at Debian 11 (meson)

fraktal-sb, 2021-10-04:
XCSoar versions having and not having the problem
-------------------------------------------------

OK 7.18
NOK 7.19

Expected behavior
-----------------

compile should end without errors

Actual behavior
---------------

see the attached copy of messages when compilig KOBO version

Steps to reproduce the behavior
-------------------------------

make -j5 TARGET=KOBO DEBUG=n  output/KOBO/KoboRoot.tgz

[7.19.err.txt](https://github.com/XCSoar/XCSoar/files/7277473/7.19.err.txt)

Do you have any idea what may have caused this?
-----------------------------------------------

missing librarys

Do you have an idea how to solve the issue?
-------------------------------------------

adding "meson" did not resolve the problem

MaxKellermann, 2021-10-04:
Does `/home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-gcc` exist?

schmidi, 2021-10-04:
Try the fixes from the following PR: https://github.com/XCSoar/XCSoar/pull/739

fraktal-sb, 2021-10-04:
/home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-gcc
exists
[image]

fraktal-sb, 2021-10-04:
Compile OK with change in meson.py (https://github.com/XCSoar/XCSoar/pull/739)

closed 2021-10-15 by MaxKellermann