XCSoar/XCSoar #775: True Airspeed calculation wrong

dspreitz, 2021-11-05:
While standing (no movement, speed = 0km/h) at the start grid, XCS showed a TAS of 26 km/h.

Issue was encountered while standing in the start grid at Jesenik (LKMI). Aerodrome Altitude = 419m
XCS is fed IOIO box from a CAI302 vario.

Order of NMEA inputs:
A) Build in GPS & Sensors
B) FLARM
C) KRT2
D) VEGA (for T and RH)
E) CAI 302
F) NANO Logger

XCSoar versions having and not having the problem
-------------------------------------------------
I tested this with Version 7.20.
I don't know if previous versions also have this issue.

Expected behavior
-----------------
When the glider is standing in the start grid, TAS should be 0 km/h.

Actual behavior
---------------

While standing in the start grid, XCS showed a TAS of 26km/h. See picture.
[image]

Steps to reproduce the behavior
-------------------------------

See attached [NMEA log](https://drive.google.com/file/d/1QyURi8HAE6q_YuJUz7xstOdL2X5PuNqD/view?usp=sharing)

Do you have any idea what may have caused this?
-----------------------------------------------

Either it is faulty input data from the vario, or a bug in the TAS calculation.

Do you have an idea how to solve the issue?
-------------------------------------------

1. Analyse NMEA log to check if the vario reported false speed data, which was then calculated into a TAS.
2. Double check TAS calculation methodt

lordfolken, 2021-11-05:
Input configuration:
XCSoar threats inputs higher up i.e. A with higher priority than B. 

So with the current configuration your IGC Pressure probe for altitude from your Flarm will be used as baro alt. 

I would put: 
A) CAI302 (Pressure including TAS)
B) FLARM (GPS, FLARM, Pressure secondary)
C) VEGA (T & RH) (Don't know this device)
D) NANO (GPS, Pressure)
E) KRT2 (Peripherial device, position doesn't matter)  
F) Built in GPS and Sensors (Backup)

As to your TAS problem:
I found this in our archives:
"log a NMEA file ("raw logger") and write a ticket. Then we'll be able to run our algorithms on that file, and reproduce exactly what it showed you on your flight. Note that the CAI302 needs to calibrate its airspeed/temperature probe after switching on. You should switch the 302 on, let it calibrate, and then power-cycle it before launching."

https://github.com/XCSoar/XCSoar/blob/master/src/Device/Driver/CAI302/Parser.cpp

/*
!w,<1>,<2>,<3>,<4>,<5>,<6>,<7>,<8>,<9>,<10>,<11>,<12>,<13>*hh<CR><LF>
<1>  Vector wind direction in degrees
<2>  Vector wind speed in 10ths of meters per second
<3>  Vector wind age in seconds
<4>  Component wind in 10ths of Meters per second + 500 (500 = 0, 495 = 0.5 m/s tailwind)
<5>  True altitude in Meters + 1000
<6>  Instrument QNH setting
**<7>  True airspeed in 100ths of Meters per second**
<8>  Variometer reading in 10ths of knots + 200
<9>  Averager reading in 10ths of knots + 200
<10> Relative variometer reading in 10ths of knots + 200
<11> Instrument MacCready setting in 10ths of knots
<12> Instrument Ballast setting in percent of capacity
<13> Instrument Bug setting
*hh  Checksum, XOR of all bytes
*/

Looking through the log for that field 7 i never see 0, but values that are siginitfanctly higher than that: 1006 for example whihch would be around 36km/h. So your vario isn't suppling the correct data.

closed 2021-11-05 by lordfolken

dspreitz, 2021-11-07:
Thank you @lordfolken for the analysis and suggestions.
I modified the order of inputs according to your suggestions.
I did indeed NOT powercycle the vario prior to takeoff. Added this item to my pre-takeoff-checklist for the future.