XCSoar/XCSoar #776: Location authorization "always" no longer possible in Android 12

btotherunner, 2021-11-08:
Hello everyone,

Unfortunately, with the update to Android 12, the "always allow" authorization no longer works.

It is currently only possible to authorize the location while it is in use.

This bug was actually fixed in the last version under Android 11 - unfortunately it is again in the current Android version 12.

Pixel 6 and 4a - Android 12 
XC Soar v.7.20-Android

Greetings from Landau

MaxKellermann, 2021-11-08:
This is due to https://forum.xcsoar.org/viewtopic.php?f=12&p=22615
This will not be solved until somebody volunteers to help.

btotherunner, 2021-11-08:
Ich helfe dir gern :D
Ist so ein tolles Projekt - wäre doch schade wenn es an solchen Punkten hängt - ich schreib dir ne Mail :D

Grüße

closed 2021-11-26 by MaxKellermann