XCSoar/XCSoar #777: Command STARTUP_REAL in default.xci not work in 7.xx versions

Vantuilmv, 2021-11-08:
I configured a command for my hardware to know when XC-Soar started, I included the following command:

mode=default
type=gce
date=STARTUP_REAL
event=SendNMEAPort1 START

------------------------------------------------

What version of XCSoar are you using when you see the issue?

This doesn't work on versions 7.xx

Which version of XCSoar is the last you know of that didn't have the problem?

It worked perfectly on 6.xx

Expected behavior
-----------------

XC-Soar sends the "START" command over the serial port as soon as I click on FLY and my hardware knows it can send data over the serial port;

real behavior
---------------

Nothing happens, this command is no longer executed;

Steps to Reproduce Behavior
-------------------------------

Just change the default.xci file and configure it as follows:

mode=default
type=gce
date=STARTUP_REAL
event=SendNMEAPort1 START

Do you have any idea what might have caused this?
----------------------------------------------

I believe that at some point in the updates someone disabled the command and forgot to enable it again or
was removed.

Do you have an idea how to solve the problem?
-------------------------------------------

Insert/fix this command so that it continues to work the same as version 6.xx.

MaxKellermann, 2022-01-07:
STARTUP_REAL works just fine, but SendNMEA did not. Fixed by a80a3949cef5990b25afcb7208e1908a89edddff

closed 2022-01-07 by MaxKellermann

Vantuilmv, 2022-01-17:
Hello, thanks for the feedback, I'm testing on version 7.21 but whenever I include the line:
event=SendNMEAPort1 START

XCSoar restarts on click fly, any other command works except SendNMEAPort1, isn't this change in version 7.21?

MaxKellermann, 2022-01-17:
> XCSoar restarts on click fly, any other command works except SendNMEAPort1, isn't this change in version 7.21?

I have no idea what that means.

Vantuilmv, 2022-01-17:
I use the default.xci file to send some commands to my device via the serial port, this command would be the takeoff command so that it emits a beep, informing that the flight has started, but when I include the event with the STARTUP_REAL, the problem occurs, the xcsoar restarts and returns to the FLY/SIM splash screen

MaxKellermann, 2022-01-17:
What is default.xci?
What is your device?
What is a "takeoff command"?
What emits a beep?
Informing whom?
What problem occurs?
Who restarts XCSoar?

Vantuilmv, 2022-01-17:
What is default.xci?
[https://github.com/XCSoar/XCSoar/blob/master/Data/Input/default.xci](url)

What is your device?
Kobo Glo HD

What is a "takeoff command"?
is a commands of xcsoar default.xci file

What emits a beep?
My device, arduino, receive data from xcsoar and emits a beep or vario sound;

Informing whom?
My device.

What problem occurs?
xcsoar restarts

Who restarts XCSoar?
himself

MaxKellermann, 2022-01-17:
This still doesn't make sense. You "send some commands to" your "Kobo Glo HD" .... what??

Vantuilmv, 2022-01-17:
I built a variometer with gps using arduino, it is connected to kobo hd through the internal serial port, so I changed the default.xci file so that kobo (xcsoar) can send commands to my device and according to the command it emits a specific sound .

Vantuilmv, 2022-01-17:
This is my file.
[xvario_old.zip](https://github.com/XCSoar/XCSoar/files/7882951/xvario_old.zip)