XCSoar/XCSoar #781: User.cup file waypoints not showing in menu.

RossDouglasMc, 2021-11-11:
XCSoar versions having and not having the problem
-------------------------------------------------
Running V7.20    V7.19 also has the problem. I haven't found one not showing the problem.

Expected behavior
-----------------
Created waypoints should be selectable in the waypoints menu when creating or modifying a task but they aren't showing.

Actual behavior
---------------
The waypoints do not show in the menu when creating a task or ammending the current task, and are therfore not selectable. If I create a new waypoint in the waypoint editor, that shows and is selectable, but the previously created waypoints are wiped from the user.cup file.

Steps to reproduce the behavior
-------------------------------
Just try to insert one of your waypoints.

Do you have any idea what may have caused this?
-----------------------------------------------
No, I jsut noticed it tonight when I tried to create a task using my waypoints that were previously useable. These waypoints are also not showing on the map any longer.

Sorry, no idea. I have not used these waypoints for a while. I tired transferring the user.cup file from my phone to my laptop, but the same thing happens on the laptop.

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2021-11-14:
I tried this with v7.20 in windows. Adding user waypoints via marker option, seems to work fine. They also show up on the map, and edititing them saves them to user.cup. But user cup isn't loaded on startup...

RossDouglasMc, 2021-11-14:
HI Philipp, Thanks for the info. I don’t know what I’ve done to get this. If I open the user file in Notepad to look at what is there then close notepad, XC Soar will no longer show the waypoints. The file is still there, and has the waypoints, but they don’t come up on the menu and don’t show on the map. If I create a new waypoint via the Waypoint Editor it deletes my previously saved waypoints form the user file. I had about 50 waypoints in the file, which I can recreate anew via the waypoint Editor, but it is a lot of work. I tried to cut and paste, in Notepad, my waypoints into another file, which will show as a selectable file, but the waypoints won’t show. I guess I’ll just have to start a new user file via the Waypoints Editor and be careful not to interfere with it at all. Although I can’t remember interfering with this one.  Anyway, thanks for looking into this. Everything else is working fine so it must have been something I did. CheersRossFrom: Philipp WollschlegelSent: Monday, 15 November 2021 12:25 AMTo: XCSoar/XCSoarCc: RossDouglasMc; AuthorSubject: Re: [XCSoar/XCSoar] User.cup file waypoints not showing in menu. (Issue #781) I tried this with v7.20 in windows. Adding user waypoints via marker option, seems to work fine. They also show up on the map, and edititing them saves them to user.cup. But user cup isn't loaded on startup..

RossDouglasMc, 2021-11-14:
HI Philipp, Thanks for the info. I don’t know what I’ve done to get this. If I open the user file in Notepad to look at what is there then close notepad, XC Soar will no longer show the waypoints. The file is still there, and has the waypoints, but they don’t come up on the menu and don’t show on the map. If I create a new waypoint via the Waypoint Editor it deletes my previously saved waypoints and starts a new user file. I had about 50 waypoints in the file (which I have a backed up copy of), which I can recreate anew via the waypoint Editor, but it is a lot of work. I tried to cut and paste, in Notebook, my waypoints into another file, which will show as a selectable waypoint file, but the waypoints won’t show. I guess I’ll just have to start a new user file via the Waypoints Editor and be careful not to interfere with it at all. Although I can’t remember interfering with this one.  Anyway, thanks for looking into this. Everything else is working fine so it must have been something I did. CheersRossFrom: Philipp WollschlegelSent: Monday, 15 November 2021 12:25 AMTo: XCSoar/XCSoarCc: RossDouglasMc; AuthorSubject: Re: [XCSoar/XCSoar] User.cup file waypoints not showing in menu. (Issue #781) I tried this with v7.20 in windows. Adding user waypoints via marker option, seems to work fine. They also show up on the map, and edititing them saves them to user.cup. But user cup isn't loaded on startup...

MaxKellermann, 2021-11-15:
@RossDouglasMc please disable HTML emails.

RossDouglasMc, 2021-11-16:
Hi Max, Sorry, I have Googled this all ways I can and can’t find a way of disabling HTML emails on either my phone or Surface Pro, which are the 2 devices I use XC Soar on. Don’t worry about it. I must have done something wrong somewhere. A mate has his running normally on a similar set up and with V.20. Mine is otherwise working fine, so please close the issue. Sorry to bother you. Best regards Ross From: Max KellermannSent: Monday, 15 November 2021 8:17 PMTo: XCSoar/XCSoarCc: RossDouglasMc; MentionSubject: Re: [XCSoar/XCSoar] User.cup file waypoints not showing in menu. (Issue ***@***.*** please disable HTML emails.

JoeggiCH, 2022-01-04:
The format of the line, which xcsoar 7.21 adds to user.cup seems be the problem. It writes the latitude into column #3 (should be column #4). I am able to fix the lines by adding an empty column to the file manually.

RossDouglasMc, 2022-01-05:
Hi,  Thank you so much. This has saved me a heap of time and solved the problem for me. I had saved the original User file, so after your email below, I changed the name of the file, went through and put an extra comma in each line to make a column 3 and it all works again. I have my 150 odd created waypoints back and useable. Thanks again, really appreciate it. CheersRoss From: JoeggiCHSent: Wednesday, 5 January 2022 2:33 AMTo: XCSoar/XCSoarCc: RossDouglasMc; MentionSubject: Re: [XCSoar/XCSoar] User.cup file waypoints not showing in menu. (Issue #781) The format of the line, which xcsoar 7.21 adds to user.cup seems be the problem. It writes the latitude into column #3 (should be column #4). I am able to fix the lines by adding an empty column to the file manually.

JoeggiCH, 2022-01-05:
I confirm the observation of Ross that 7.18 works and 7.19 does not work (all Win64).

Example data from user.cup:
```
"Working - created with 7.18","718",,4607.175N,01034.825E,2790M,1,,,,""
"Broken - created with 7.19","719",4607.175N,01034.825E,2790M,1,,,,""

```

lordfolken, 2022-01-05:
Looks like the country field is missing. 
https://downloads.naviter.com/docs/SeeYou_CUP_file_format.pdf

What i dont yet get is that it is in the code:

https://github.com/XCSoar/XCSoar/blob/464df291983948c67e967725e613b25c3efd9814/src/Waypoint/CupWriter.cpp#L121

commit 12d53588 (2022-01-05): cupwriter: add missing comma for shortname
fixes #781
PR #826 (2022-01-05): cupwriter: add missing comma for shortname
fixes #781
closed 2022-01-05 by MaxKellermann