XCSoar/XCSoar #782: XCSoar not saving igc file since version 7.xx

andymel2, 2021-11-17:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

andymel2, 2021-11-17:
Would be very grateful if version 7.xx would save the igc file!

andymel2, 2021-11-17:
Did not mean to close it!

lordfolken, 2021-11-17:
There isn't even the most basic information in this bug report. No one can help you like this.

andymel2, 2021-11-17:
This was not an empty report when I wrote it!  All the above sections were completed.

However, it is quite a basic problem.  For XCSoar 6.xx (PC Version), the .igc file was automatically saved at the completion of a task; it was to be found in XCSoarData/Logs and was still there after shutting down XCSoar.

Since I started using 7.xx, I can see the .igc file in the logs folder while the task is running, but as soon as I end the task and shut down XCSoar, the .igc file disappears from the logs folder.  I have searched my PC to see if it is being moved elsewhere - no.

I only use the PC version of XCSoar.

The igc file is, of course desirable for flight analysis.

lordfolken, 2021-11-18:
Ok, i could reproduce this with the WIN64 target with the current HEAD.
The file is indeed deleted as soon as XCSoar exits

lordfolken, 2021-11-18:
```
[2021-11-18T17:27:06Z] Starting XCSoar X
[2021-11-18T17:27:06Z] Display dpi=96,96
[2021-11-18T17:27:06Z] Initialise fonts
[2021-11-18T17:27:07Z] Startup dialog
[2021-11-18T17:27:07Z] Loading profiles
[2021-11-18T17:27:07Z] Loaded profile from C
[2021-11-18T17:27:07Z] Units: GetUserDefaultUILanguage() = 0x809
[2021-11-18T17:27:07Z] Loading language file
[2021-11-18T17:27:07Z] Language: GetUserDefaultUILanguage()=0x809
[2021-11-18T17:27:07Z] Loading input events file
[2021-11-18T17:27:07Z] LoadTerrain
[2021-11-18T17:27:07Z] Loading Topography File...
[2021-11-18T17:27:07Z] ReadWaypoints
[2021-11-18T17:27:07Z] Failed to read waypoint file: C
[2021-11-18T17:27:07Z] Failed to read waypoint file: waypoints.xcw
[2021-11-18T17:27:07Z] Failed to open 'airfields.txt' from ZIP file
[2021-11-18T17:27:07Z] RASP load
[2021-11-18T17:27:07Z] No rasp data file: Failed to open ZIP archive C:\users\folken\My Documents\XCSoarData\xcsoar-rasp.dat
[2021-11-18T17:27:07Z] ReadAirspace
[2021-11-18T17:27:07Z] Failed to load airspaces from map file: Failed to open 'airspace.txt' from ZIP file
[2021-11-18T17:27:07Z] Register serial devices
[2021-11-18T17:27:08Z] ProgramStarted
[2021-11-18T17:27:08Z] cannot open C:\users\folken\My Documents\XCSoarData\lua\init.lua: No such file or directory
[2021-11-18T17:27:08Z] Start at terrain center
[2021-11-18T17:27:08Z] Opening device T
[2021-11-18T17:27:39Z] Logger Started: C
[2021-11-18T17:27:52Z] Entering shutdown...
[2021-11-18T17:27:52Z] Logger stopped: C
[2021-11-18T17:27:52Z] Stop threads
[2021-11-18T17:27:52Z] Waiting for calculation thread
[2021-11-18T17:27:52Z] Waiting for draw thread
[2021-11-18T17:27:52Z] delete MapWindow
[2021-11-18T17:27:52Z] Save default task
[2021-11-18T17:27:52Z] Stop COM devices
[2021-11-18T17:27:52Z] Close Progress Dialog
[2021-11-18T17:27:52Z] Close Windows - main
[2021-11-18T17:27:52Z] Finished shutdown
```

lordfolken, 2021-11-18:
The missing ':' after the c: is suspicous.

lordfolken, 2021-11-23:
So this cost me 3 evenings.. I learnt quite a bit about gdb, strace and c++. 
In the end I found the function that deletes the file. It was duplicated between windows and the rest of the platforms. 
Some logic with exceptions was added over time, that is why it worked on the Linux side. the Logic never made it to the windows variant of the function, which is why the file was always deleted without exception. 

My proposal merges the two functions and only uses the platform specific deletion call.

closed 2021-12-08 by lordfolken