XCSoar/XCSoar #783: 3 InfoBoxes: More information about thermaling efficiency

bomilkar, 2021-11-19:
XCSoar version
--------------
7.*

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
I’m proposing to augment 3 InfoBoxes to better judge the quality of thermals. In all 3 cases I can use the comment line to display additional information about thermals. It helps answering questions like "am I spending too much time in this thermal", "should I continue circling", "how much time did I invest in the last thermal".

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
The following 3 InfoBoxes are affected:
“TL Gain” shall include the overall vertical velocity (e.g. climb rate) of the last thermal.
“TC Gain” shall include the time spent in the current thermal since circling started, up to now.
“TC 30s” shall include the climb rate of the current thermal since circling started, up to now.

“TL Gain”:
Last thermal gain. Total altitude gain/loss.
**Adding** time spent in last thermal.

“TC Gain”:
Thermal gain. The altitude gained/lost.
**Adding** climb rate in the current thermal.

“TC 30s”:
Thermal climb, last 30 s.
A 30 second rolling average climb rate based of the reported GPS altitude, or vario if available.
**Adding** the time spent circling in the current thermal.

I’m planning a PR with 3 commits to address the above.
Any objections?

bomilkar, 2021-11-19:
This is how it should look.

[image]
[image]
:

I tested it during the last season and found it quite useful.

mlep, 2021-11-19:
Nice propositions of improvement.

A suggestion for TC Gain, so that it is easier to read: time in `minutes' seconds"` so that `154 s` is displayed as `2' 34"`.
And to help answering "should I continue circling" and "how did I do", could you use color in the display:
- for TC 30s: Vz in red since 2.7 < 3.1. And Vz in red if < 0.
- for TC Gain and TL Gain: height in red if loss of height (=negative gain)
- for TL Gain: mean Vz in red if < 0.

lordfolken, 2021-11-19:
This makes total sense to me. Any volunteers for a PR?

lordfolken, 2021-11-19:
Please don't use red. Red signifies emergency.

bomilkar, 2021-11-19:
Thanks guys for the feedback.
I can work on the PR. I have working/tested patches already. No big deal at all. I'm planning to change the code and add to the help text. I think the manual doesn't need any change.
@lordfolken do you think the PR should consist of **3 commits** or **one single commit** for the 3 InfoBoxes.

@mlep you're suggesting color coding. I personally don't feel there is much merit to this. To me color coding can be confusing particularly if I don't understand the scheme, WRT @lordfolken comments rightly so: red signifies emergency. Probably green and black?

However, if we decide to color code, then the scheme should be consistent with the main value's color code in that particular infobox, In terms of value threshold and color. E.g. _current_risk_mc_ as the threshold for TC 30s.

For the duration in the "TC Gain" InfoBox: I don't like the mm:ss notation. That is because I use the 2 values (gain and duration) in different ways:
gaining 500m in 250sec is not so bad.
gaining 250m in 500sec can be a wast of time
gaining 20m in 10sec is not relevant.
I hope you understand how I'm using these 2 numbers. For that the "raw" seconds figure is more useful.

bomilkar, 2021-11-19:
Thanks for pointing to the color scheme as documented in the Development Manual.
So in my opinion for the "TC 30s" InfoBox we should pick black and green. More than anything else it is "Green for positive indicator of safety" if we're climbing well. That means:
**Green** as soon as the climb rate exceeds (MacCready * safty_factor).
**Black** otherwise.

FrederikVSK, 2021-11-19:
On the picture, what does the +3.1m/s indicate? Last thermal total or current thermal total...and is your plan to swap that out with time spent circling in the current thermal?

“TC 30s”:
Thermal climb, last 30 s.
A 30 second rolling average climb rate based of the reported GPS altitude, or vario if available.
Adding the time spent circling in the current thermal.

/FrederikDK

bomilkar, 2021-11-19:
My mistake. Sorry!
Related to the picture with the "TC 30s" InfoBox. The +3.1m/s value in the comment line is the climb rate since the start of the thermal, GPS based or vario based whichever is available.
The main value in that IB is the "30 second rolling average climb rate based of the reported GPS altitude, or vario if available". That shall be uncanged.

BTW: The more I think about the color coding of the numbers in these 3 InfoBoxes the more I'm convinced **color coding makes no sense**. Color coding has the potential to confuse and distract the pilot unless there is a clear meaning and a well established scheme.

lordfolken, 2021-11-19:
I think you can leave the color coding away for now.
There would be a global switch to turn all info boxen bicolor (w/b or b/w). (colored infoboxes) in the settings.

To your commit question:
 * Please do 3 commits, but each one must build and be atomically complete. 
 * Please be very clear on your commit message
 * If your commit requires more text than available in the description you can use the git commit summary
 * rebase your commits before pushing

lordfolken, 2021-11-19:
Btw i just saw, the tc is currently colored red when:
https://github.com/XCSoar/XCSoar/blob/e8d9ca024852ddefc86cec22d56a6c02a039b667/src/InfoBoxes/Content/Thermal.cpp#L191
  data.SetValueColor(CommonInterface::Calculated().task_stats.total.vario.get_value() < 0 ? 1 : 0);

MaxKellermann, 2021-12-23:
PR was closed by submitter because he lost interest in finishing it, so let's close this feature request as well.

closed 2021-12-23 by MaxKellermann