XCSoar/XCSoar #789: ttymxc0 should be listed as a port, but it's not

FogWatch, 2021-12-04:
XCSoar versions having and not having the problem
-------------------------------------------------
Kobo Clara running 4.30.18838. BlueFlyVario_TTL_GPS_v11
XCSoar 7.20

Expected behavior
-----------------
ttymxc0 should be listed as a port option. ttya4, ttya5 etc are listed, but not ttymxc0.

Actual behavior
---------------
ttymxc0 is not listed as a port, which means that I cannot connect the BlueFly. 

When running Nickel 4.30.18838:
ls -la /dev/ttymxc0
crw-rw----    1 root     root      207,  16 Dec  3 11:41 /dev/ttymxc0

Steps to reproduce the behavior
-------------------------------
Config, Devices, Edit, Port: ttymxc0 should be listed as an option, but it's not.

Do you have any idea what may have caused this?
-----------------------------------------------
6.8.17 did not have this problem. I'm guessing that the problem might be related to changing from "/bin/mknod /dev/ttymxc0 c 207 16" to "/sbin/udevd".

Do you have an idea how to solve the issue?
-------------------------------------------
No

lordfolken, 2021-12-04:
Can you do an ls -l on 6.8.17? could be a permission issue, and unless xcsoar runs as root, it won't be able to access 
```crw-rw---- 1 root root 207, 16 Dec 3 11:41 /dev/ttymxc0```

FogWatch, 2021-12-04:
"Can you do an ls -l on 6.8.17?"
No. I can only do things like that running Nickel. It's only in there that I have wifi and can telnet.

nelliott, 2021-12-14:
Can confirm this on 7.21

```
~ # ls -alh /dev/ttymxc0 
crw-rw----    1 root     root      207,  16 Dec 14 13:00 /dev/ttymxc0
```

FogWatch, 2021-12-25:
This is the sequence.

On my Kobo Clara, 6.8, BlueFly: 
Config, Devices, Edit, Port, ttymxc0 was listed and functioned fine.

With 7.20, ttymxc0 was not listed and there was no signal from the BlueFly. So, I reverted to 6.8.

Under 7.21, ttymxc0 is still not listed under Port but there is signal from the BlueFly and location and vario are resolved:

```
# cat /mnt/onboard/XCSoarData/xcsoar.log | head -n 1
[2021-12-25T04:31:12Z] Starting XCSoar XCSoar v7.21-Kobo
```
```
# ls -alh /dev/ttymxc0
crw-rw----    1 root     root      207,  16 Dec 25 04:33 /dev/ttymxc0
```
I'm loving the ability to telnet into my XCSoar, without having to patch or go into Nickel.

I think this can be closed.

closed 2021-12-25 by FogWatch

MarkusPethe, 2022-03-27:
I have the same issue running XCSoar 7.23
Please help me to understand how I can solve this issue,

lordfolken, 2022-03-28:
@FogWatch can you help?