XCSoar/XCSoar #795: .cupx file format

hjr, 2021-12-11:
What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

I'd like to see the naviter cupx waypoint file format supported as we do support the cup file format. The 'x' additions are columns for a description and a list of picture file names. jpg picture files are embedded in the cupx file like in our xcm tar files. 

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

Drop in cupx files to the XCSoarData folder from the glider pilot community and just use them, incl. the pics feature for outlanding fields etc.
(just wrote a python script to create the Airfield-Details.txt glue file to make use from an cupx file)

MaxKellermann, 2021-12-11:
This has been several times before, and in the end, this was impossible because the file format is proprietary, there exists no public specification, and nobody was ever willing to reverse-engineer it. So unless you have a public and free specification, we need to close this feature request.

hjr, 2021-12-12:
I am willing to give a reverse engineered spec a trial.

Plantain, 2021-12-12:
It's nothing particularly complex, just two .zip files concatenated - it
was first reverse engineered in a prior XCSoar ticket (now deleted), but
you can see snippets of it still
https://tophatsoaring.atlassian.net/browse/TH-760

MaxKellermann, 2021-12-12:
Why... would anybody ever concatenate ZIP files?? It's already a container format.

lordfolken, 2021-12-12:
Landewiesen.at is available as cupx. They must have some kind of implementation.

hjr, 2021-12-15:
Will try to connect to them.

hjr, 2021-12-20:
Christian Hynek (chynek@streckenflug.at) answered, you guys seem to have contact as well. 
So the plan is to create an handsome and open format to distribute landewiesen.at from its proper origin and have XCSoar enabled to read those format too.
You may use this thread to post updates on this.

lordfolken, 2022-01-04:
So i got a mail from naviter, that they updated the documentation and also described cupx format:

Documentation for CUP and CUPX file formats is updated now.
You can download it here:
. for CUP: https://downloads.naviter.com/docs/SeeYou_CUP_file_format.pdf
. for CUPX: https://downloads.naviter.com/docs/SeeYou_CUPX_file_format.pdf
Have a nice 2022 intro,

MaxKellermann, 2022-01-04:
This format is so braindead, I will not waste time with writing code to support it.
If this were a single ZIP file containing the CUP file and the images, okay ... but two ZIP files, the latter of which only contains the CUP file and nothing else - seriously??

scls19fr, 2022-01-09:
I wonder if splitting a cupx file into 2 zip files can also be done in just a command line under Windows and Linux. I would be very curious to see how it can be done. I understand that cupx idea was a kind of [NIH syndrome](https://en.wikipedia.org/wiki/Not_invented_here) avoidance 😀

MaxKellermann, 2022-01-10:
The badness of CUPX is that it is two ZIP files instead of one. How is this NIH avoidance? It could have been so damn easy, the most obvious solution would be so trivial to implement (just one ZIP file containing everything), but they chose to make life hard, for no reason.

scls19fr, 2022-01-10:
Maybe we should define our own format as suggested by @lordfolken in https://github.com/XCSoar/XCSoar/issues/745#issuecomment-940171944 and try to make it a "new standard" for waypoints with details (details beeing pictures and/or pdf documents). But I wonder what will be position of Naviter and landewiesen.at about such a format.

https://xkcd.com/927/

lordfolken, 2022-01-10:
So the fun part is:
If you use unzip on the file you get the first archive.
If you use 7z you get the second archive. 

7z starts from the back of the file whereas unzip seeks from the beginning.

scls19fr, 2022-01-10:
Isn't there a way / a cli tool to detect "magic number" or any other data marking either start of zip file or end of zip ?

I noticed with a hex editor that

    50 4B 03 04 14 00 00 00 08 00 A7 4C 77 4B

can be found 3 times in `small_sample_file.cupx` and 

    50 4B 03 04 14 00 00 00 08 00 7C 4E B2 4A

can be found more othen in `big_sample_file.cupx`

scls19fr, 2022-01-10:
I'm not skilled enough at playing with [Matryoshka (-M)](https://en.wikipedia.org/wiki/Matryoshka_doll)

    binwalk --dd='.*' small_sample_file.cupx -M

maybe some one is better at that task or using an "[hachoir](https://fr.wikipedia.org/wiki/Hachoir)" https://hachoir.readthedocs.io/

lordfolken, 2022-01-11:
i have written a converter that generates a waypoint_details file from the extracted data.

lordfolken, 2022-01-11:
https://github.com/lordfolken/xcsoar-cupx-converter/blob/master/xcsoar-convert-cupx.py

I'll try to incoperate foremost to make it easy to use.

scls19fr, 2022-01-11:
A better solution could be to use hachoir or binwalk because they are (or have) python module (but I haven't try myself). So it could avoid external dependency which may be problematic especialy for Windows users.

lordfolken, 2022-01-11:
so i implemented now with binfmt. It finds the files and extracts them. 
basically if you copy the output dir into your XCSoarData it will work.

some thoughts:
there is more bulletproofing required. I found several files with different 'accents'. 
Potentially python-areofiles should be used to read / write the cup files. 
Tasks are currently not handled, but are just copied.
Redudant information should be removed. The fields with comments + pics should not be in the 
resulting cup file, only in waypoints_details.

We have to think about a delivery format. Our filemanager basically only supports single files for download. Plus handing the user a couple of files that he has to copy to the XCSoarData dir is also an increasingly difficult proposition. (People no longer think in files/dirs, Android is actively destroying file access as a concept, ease of use.)

So i was thinking to do it the same way along the xcm file. A single zip file that is _not_ compressed. So it can be used to seek the files inside. The size difference should be small, as jpegs are already compressed, and the cup file is easily zipped with the stream compression of the serving webserver.

MaxKellermann, 2022-01-11:
We should better have a converter which converts those concatenated ZIP files into just one ZIP file. The rest I'll implement in XCSoar. Our waypoint_details is a horrible crappy design (nearly as bad as CUPX).

lordfolken, 2022-01-11:
I can adapt my script to convert it all to one zip. 

I agree that wp_details isn' t that well designed. I guess it had to be easy for the time it was designed in and is an ill extension to cup as cupx is to cup.

We can always implement aixm: https://github.com/aixm/donlon/blob/master/EA_AIP_DS_FULL_20170701.xml ;-)

scls19fr, 2022-01-11:
Maybe we should define a format for that purpose.

I personaly like the concept of DataPackages https://dataprotocols.org/data-packages/ with description in a RFC manner.
https://specs.frictionlessdata.io/ (it's JSON based... not XML based !)
CUP file could be considered as a tabular data package https://specs.frictionlessdata.io/tabular-data-package/
There is several languages supported Python, R, Java, Javascript, Ruby, Julia, Go... but neither C nor C++ according https://libraries.frictionlessdata.io/docs/data-package/python/

So a more simplistic format should probably be considered.
zip file (named for example .zcup... that's just an example) containing at least
version.json (or .yml or .toml or .ini) which contains at least version of zcup version (semantic version of file version could probably be considered)
points.cup a cup file containing at least one waypoint as describe by https://downloads.naviter.com/docs/SeeYou_CUP_file_format.pdf
optionaly
a `wpt` directory with inside, optionaly, directories with waypoint names which could contains JPG, PNG image and PDF documents
but such a structure may encounter problems on some file systems because of

MaxKellermann, 2022-01-11:
Don't make it too complex and complicated.
We have the XCM format already, and it can already contain a CUP file (and an airspace file). Just specify how per-waypoint images/files can be stored (similar to CUPX), and optionally omit topography/terrain, and we're done.

lordfolken, 2023-03-22:
A design proposal for an archive format like XCM is here #1025

antonka2000, 2023-05-30:
Up until Christan Hynek is giving the files in a useable format, I used this approach:

Install maybe not installed programms:

apt-get install binwalk unzip

Load CUPX and CUP file from streckenflug.at .
The .cup file can be used directly in xcsoar.
The .cupx file has to be prepared:

binwalk 20230404_zentral_und_ostalpen_de.cupx

at the end you see something like this:
38117316      0x2459FC4       End of Zip archive, footer length: 22
38117338      0x2459FDA       Zip archive data, at least v2.0 to extract, compressed size: 22029, uncompressed size: 72991, name: POINTS.CUP
38139463      0x245F647       End of Zip archive, footer length: 22

extract form the beginning of the file to the beginning of POINTS.ZIP :
( head -c <offset> Afile > irgendwas.zip )

head -c 38117338 20230404_zentral_und_ostalpen_de.cupx >mypics.zip

then:

unzip mypics.zip

now you see the folder “Pics/” with the pictures inside.

You get the waypoints by adding 1 to the End-Value of “head”:
( tail -c +<offset+1> Afile > nochirgendwas.zip )

tail -c 38117339 20230404_zentral_und_ostalpen_de.cupx >points.zip

unzip points.zip

Now you get POINTS.CUP

Out of this file you have to create a file like this: 

[FROJ4912]
image=pics/1_4912_osm.jpg
image=pics/2_2880.jpg

Open the file with the editor vim:

vi POINTS.CUP

use the following commands:

dd			delete the first line
:%s/^[^,]*,\s*//	delete everything from the beginning of a line to the first ,
:%s/,.*,/,/		delete everything from the first to the last , 
			and replace it with a ,
:%s/"//g 		delete all “

now it should look like this:

ABFA0165,1_165_osm.jpg;2_186.jpg
ACHE0380,1_380_osm.jpg;2_748.jpg
ACHE0166,1_166_osm.jpg;2_191.jpg

now continue with:

:%s/^/[/  			put a [ in front of each line
:%s/,/]\rimage=Pics\//	replace each , by a ], a “CR” and "image=Pics/"
:%s/;/\rimage=Pics\//   replace each ; by a “CR” and "image=Pics/"

:w waypointdetails.txt   save with this name

Now the file should look like this:

[ABFA0165]
image=Pics/1_165_osm.jpg
image=Pics/2_186.jpg
[ACHE0380]
image=Pics/1_380_osm.jpg
image=Pics/2_748.jpg

The folder “Pics” and the file “waypointdetails.txt” have to be stored in the data folder of xcsoar. - And of course be selected in xcsoar.

llauner, 2024-01-08:
> Now the file should look like this:
> 
> [ABFA0165] 
> image=Pics/1_165_osm.jpg 
> image=Pics/2_186.jpg

I tried to add multiple images for a given point but it doesn't really work:
When you click on the right arrow you navigate through the details page. At one point it'll show you image 1.
To see image 2 you have to navigate one step right and one step back to the left !
It's totally not natural at all
I was expecting to see image 1, then image 2 and backward when clicking on left  arrow...
In other words multiple pictures is not really usable at the moment...