XCSoar/XCSoar #799: Share current position

scls19fr, 2021-12-17:
XCSoar version
--------------

7.20

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Beeing able to share position using email or SMS could be a nice feature to have.

Latitude / longitude could be used but plus codes could also be interesting

http://plus.codes

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

Use case : a glider is landed outfield and pilot want to quickly share its position to teammates.

This could be done by clicking on glider, details and adding a Share menu entry.

closed 2021-12-17 by MaxKellermann

scls19fr, 2021-12-17:
Thanks @MaxKellermann for this very quick improvement!
I'm pretty sure some glider pilots in my club will use this feature! 
Any opinion about "plus codes" aka [Open Location Code](https://fr.wikipedia.org/wiki/Open_Location_Code)?

scls19fr, 2021-12-18:
I noticed https://github.com/Ksar/XCSoar/commit/8cc3e398163fcc8ca32b3f9473dd04e8ab1e6910 add a "Share" button to Info 2/3 / Status: Flight. When clicking on glider icon on map and "Your Position" entry, Details button is disabled. What is your opinion about making this button pointing to Status page? It could help the process of sharing current position when required.

MaxKellermann, 2021-12-18:
Yes, we could enable this button.
I have no idea about these open location codes, but if they are helpful, we should have them. This is a really easy coding task - any volunteers?

scls19fr, 2021-12-18:
Not sure I can implement it in XCSoar currently (because what looks easy to you doesn't means it's easy to me) but CPP implementation of open location codes can be found at https://github.com/google/open-location-code (and various other languages)