XCSoar/XCSoar #802: ide/provisioning/install-debian-packages.sh and install-android-tools.sh saved with <CR><LF>

lpojzl, 2021-12-20:
XCSoar versions having and not having the problem: current master branch
-------------------------------------------------

Expected behavior
-----------------
when deploying build environment for building XCSoar for Kobo through Vagrant, the virtual machine shall be fully provisioned.

Actual behavior
---------------

Provisioning fails with:
"XCSoarDevEnv: sudo: unable to execute /xcsoar-host-src/ide/provisioning/install-debian-packages.sh: No such file or directory"

Steps to reproduce the behavior
-------------------------------

- clone current master branch
- cd to ide/vagrant
- execute "vangrant up"

Do you have any idea what may have caused this?
-----------------------------------------------

install-debian-packages.sh has been saved in Windows editor. The same is valid for install-android-tools.sh:

vagrant@XCSoarDevEnv:/xcsoar-host-src/ide/provisioning$ cat -v install-debian-packages.sh
#!/bin/bash^M
^M
set -e^M
^M
export DEBIAN_FRONTEND=noninteractive^M
declare -A APTOPTS^M
APTOPTS[1]="--assume-yes"^M

Do you have an idea how to solve the issue?
-------------------------------------------

Save both scripts using Unix txt file format. Or you can convert with:
tr -d '\r' < install-debian-packages.sh > install-debian-packages.sh.new

MaxKellermann, 2021-12-20:
Are you really sure this is a problem in the master branch, and not a local problem on your computer? Because over here, everything is in unix text format.

lpojzl, 2021-12-20:
It must be the case if github is clean. ;-) Feel free closing the issue. Sorry to bother!
(Got new laptop with vanilla install)

lordfolken, 2021-12-21:
Are you checking it out on a windows system? Git will checkout on windows with crl/lf.

lpojzl, 2021-12-22:
Yes. this is what happened. I was hoping to contribute actively to the project.
Hopefully soon and better. ;-)
How can I close this issue?

lordfolken, 2021-12-22:
Welcome to the club! 
I'll close it for you.

We are available to chat here: https://matrix.to/#/!xxdFVPQHJTfdSXGpFE:kabelsalat.ch?via=kabelsalat.ch&via=libera.chat&via=matrix.org

closed 2021-12-22 by lordfolken