XCSoar/XCSoar #804: XCSoar not started bi Rassbberry PI4

urb31075, 2021-12-21:
I have compiled from source from repository and am trying to run:

urb@xcsoar:~/XCSoar/output/UNIX/bin $ ./xcsoar
Starting XCSoar XCSoar v7.21-Linux~git#2495232717
Using DRI device /dev/dri/card1 (use environment variable DRI_DEVICE to override)
drmModeGetResources() failed

urb@xcsoar:~/XCSoar/output/UNIX/bin $ sudo ./xcsoar
Starting XCSoar XCSoar v7.21-Linux~git#2495232717
Using DRI device /dev/dri/card1 (use environment variable DRI_DEVICE to override)
error: XDG_RUNTIME_DIR not set in the environment.
drmModeGetResources() failed

urb@xcsoar:~/XCSoar/output/UNIX/bin $ uname -a
Linux xcsoar 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux

lordfolken, 2021-12-21:
Do you have a gui running on your rasberrypi? If so please terminate it before starting xcsoar from console.

urb31075, 2021-12-22:
Hello!
I'm a little confused.
*  I installed the desktop raspbian PI4, installed all updates.
*  I clone  https://github.com/XCSoar/XCSoar.git  and  compiled according to the instructions  https://xcsoar.readthedocs.io/en/latest/build.html
*  In raspi-config select «System Option» «Boot / Auto Login» «Console»
I run in console ./XCSoar/output/UNIX/bin/xcsoar and i see splash screen as in the photo №1
I can see the mouse cursor, but it doesn't move. Also doesn't work touchscreen.
I press Ctrl+C and program terminated.
*  In raspi-config select «System Option» «Boot / Auto Login» «Desktop»
 I run xcsoar and i see error:  ​​​​​​​ drmModeGetResources() failed
How to run the program correctly?
I expected that it would be possible to launch it in the desktop version, because before launching it is necessary to calibrate the touchscreen, launch other applications, connect to the raspberry by VNTs, etc.

> 
>Do you have a gui running on your rasberrypi? If so please terminate it before starting xcsoar from console.
 
 
С уважением, Угрюмов Роман.

urb31075, 2021-12-22:
[image]
[image]

MaxKellermann, 2021-12-22:
> How to run the program correctly?

You have figured it out already, havn't you? If your graphical desktop occupies the graphics chip, XCSoar can't run, it needs exclusive access.

closed 2021-12-22 by MaxKellermann

urb31075, 2021-12-22:
I understood, let the graphic work monopolistically. But why doesn't the mouse and touchscreen work? I move the mouse, it does not move on the screen and does not click on the buttons.