XCSoar/XCSoar #807: Split the weight config properly into plane and flight aspects.

hjr, 2021-12-22:
Currently we have a plane setup consisting out of a polar including the reference weight and a "dry mass" to adjust the given reference mass to the actual flight mass excluding all water ballast. The flight setup adds the loaded water ballast.

Issue:
dry mass is including everything but water ballast, means water, pilot(s), and potential deviation of your glider from the typical glider model empty weight. This mixes up flight and plane setup. It also forces us to provide the known PIL/PAS two seater polars, reflecting the different flight setups.
 
Suggestion:
Plane: Replace "dry mass" with "empty mass", to just cover the adjustment from reference mass to actual empty mass.
Flight: Add a "Pilot+ mass", meaning everything that is loaded to the glider and is not droppable water ballast. For motor glider this could also include fuel if needed.

PIL/PAS polars could be replaced, removed. The nature of our plane profiles would be up valued. The structure of plane flight setup enhanced.

Thx

hjr, 2021-12-26:
After looking into this a bit here my proposal. First the as is implementation:
[image]

Would then look like:
[image]
Red items change.
PolarShape, now owns the reference weight because their is no shape of a polar w/o mentioning the weight it refers to. Copies of this ref_weight removed.
Plane, got rid of the dry_mass, and got an empty_mass instead.
GlidePolar, will hold the new "crew" load info that gathers all masses load on top of the planes empty mass and is not dumpable. We might keep to copy of the original polar reference_mass here, because here we calculate the update to the polynome whenever needed.
Will go on with this in case I get positive feedback.

hjr, 2022-01-15:
What would be the preferable place to store the Pilots weight? The users Profile?

lordfolken, 2022-01-16:
One option would be to separate the Pilot Profile into a separate file. 
Then there would be
 * Pilot Profile (Name, Weight)
 * XCSoar Configuration Profile (Flight type, XC, Competition, ..)
 * Plane Profile (Weight, Ballast, Cruise Speed, Logger ID etc. )

hjr, 2022-01-18:
Good thoughts, thanks for sharing. I also had these things in mind for a while. 
The recurrent observation is actually that the personal taste on what XCS should show on what page, and what elements on the map are extremely different. So the XCS config portion is actually keeping the personality and the pilots name and weight is just a negligible detail of it. Having the pilots name and weight as a swap-able portion of the XCS profile like the plane is thus contradiction. You try to cut a personality in two parts that kind of always want to be one thing.

I could imagine that putting the Pilots name and the needed attributes like weight just to the top of a XCS profile would support the idea just fine. The logger configuration would then inherit the needed attributes and not own the pilots name any more. A very subtle change only.

No additional file not much change, a straight forward move on the profile .. what do you think?

hjr, 2022-01-22:
The idea got covered with the current implementation. Details might be raised and changed through further new issues. Closing it.

closed 2022-01-22 by hjr