XCSoar/XCSoar #812: B50 cruise / climb switch of displays now incorrect after fix

gldrpilot, 2021-12-24:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?

-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2021-12-24:
https://github.com/XCSoar/XCSoar/pull/774 this was the discussion and merge. @iltis42 can we verify?

gldrpilot, 2021-12-24:
I have double checked using the simulator, see attached file of results. This suggests the status is inverted to the published documentation as mentioned in #774 
v7.20 works in the correct sense when connected to either the B800 and the simulator.
[Cruise climb status to PBB50 string.pdf](https://github.com/XCSoar/XCSoar/files/7775149/Cruise.climb.status.to.PBB50.string.pdf)
.

gldrpilot, 2021-12-24:
Accidentally closed this...

iltis42, 2021-12-24:
Fully agree, öooks like the Borgelt documentation is wrong, exactly inverted. Hence the B50 or B800 gadgets are like this in the field, and we can't change, so we needed to revert in XCSoar in order to fix this:
https://github.com/XCSoar/XCSoar/pull/774
and i we agree this as solution, and keep the workaround of this bug in XCSoar, i will also change XCVario either to inverted cruise sense, so things will work with both gadgets in future.

iltis42, 2022-01-03:
Can we close the issue with the above reversion?
Mike Borgelt has answered my mail and confirmed by looking at the code, that documentation has twisted the sense.

gldrpilot, 2022-01-03:
Thank you, I'll close this.

gldrpilot, 2022-01-27:
Hi
I see a release of XCSoar 7.22, however the topic of this issue has not been corrected.

gldrpilot, 2022-01-27:
I see a release of XCSoar 7.22, however the topic of this issue has not been corrected.

gldrpilot, 2022-02-14:
Now release 7.23 is available but on testing this error is still in place. Any idea when it will be restored to correct sense?

Thanks -

iltis42, 2022-03-30:
Can confirm the bug is still there, somehow the fix wasn't merged. See also my comment here:
https://github.com/XCSoar/XCSoar/pull/813#issuecomment-1080001192
**Correct is bit 0 (zero) is CRUISE, and bit 1 is CIRCLING.**
I even see this change in the file-list of #813  https://github.com/XCSoar/XCSoar/pull/813/files
[image]
Hence the current state is not like this:
[image]

iltis42, 2022-04-13:
@lordfolken apparently the problem is, your fix:  https://github.com/XCSoar/XCSoar/compare/revert-774-cruiseFix
did not go into 7.23 so issue still persists. May it be possible to merge this in for the next build ?

PR #905 (2022-04-13): Revert "Invert Cruise/Circling mode in NMEA parsing of Borgelt B50 dr…
This reverts commit d1125e1f6c265365f2e511a6e16ca1efe8b64092.

Brief summary of the changes
This will revert a change that came in as of an issue in Borgelt documentation (inverted cruise bit in $PB50 NMEA protocol), and so in turn will fix issue referenced below.

Related issues and discussions
Closes #812
https://github.com/XCSoar/XCSoar/issues/812
closed 2022-04-13 by lordfolken