XCSoar/XCSoar #814: Support for UL motorflight

volkmarnissen, 2021-12-27:
What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
I am flying a pipistrel sinus. It's like a TMG usable for gliding and motor flying.
Currently I am using skydemon for motor flying and XCSoar for gliding.
Reason:
Skydemon is able to
1. Show GAFOR weather forecast in navigation view
2. Approach- and departure routes for each airport
3. Simple to use for motor flying (simple UI, no gliding information)
4. Submit flight plans (E.g. for crossing borders)

On the other hand XCSoar is open source and has more functionality
E.g. set radio frequency and all the gliding features.

I'd like to design and implement the following:
==============================
1. Approach and departure routes for each airport
 - Create a web page to get .kml files describing the approaches and departures from airport owner or pilots
 - Create a .shp file for the approaches/departures append it into .xcm file
 - Alternative: Integrate the usage of AIP pdf files in XCsoar
2. Show GAFOR
 - Create a .shp file containing the GAFOR areas
 - Show the weather condition in different colors and with labels for the areas and for the airports
3. Simple use for motor pilots
- Integrate an assistent for setup XCSoar to download and activate the required files and select the usage type motor/gliding
- The new "type" motor will contain a simple UI containing a navigation page and only a few info boxes.

I am searching for volonteers who want to support development and I am looking for feedback about this feature request.
These features can be integrated in some newer release.
First I'd like to contentrate on Approach/Departure feature
Then the weather feature
At last I'd like to simplify the setup.
Each feature should be developed in a separate branch

lordfolken, 2021-12-27:
There seems to be a airport circuit file for Germany here. 

https://www.dropbox.com/sh/4ebk4ex6407691y/AABkx2R2_7Baa_dDVQg1R9xOa/VFRNav/Version3.xx?dl=0&preview=SeglerPlatzrunden_VFRnav_v3_N_5.01.09.27.txt&subfolder_nav_tracking=1

We have PDF support in the waypoint details files.

Additional shape files could be added to the maps. 

Gafor charts could be implemented in various ways. Di you have a data source?

volkmarnissen, 2021-12-27:
@lordfolken: The link point to glider circuits rather tan motor airport circuits. Do you have a source for them also?
XCSoar could also be used for TMG. The motor circuits are more critical, because of noise emission.
Waypoint files as pdf can be an option. I will check that.
Weather data:
I know, there is a DWD API and I know, skydemon has the Gafor feature. It can be enabled after entering pc_met user name and password. So, I assume, there is a way, which I don't know yet.

So, I'll investigate
1. How to get the AIP pdf's and how to integrate them into waypoint details. This is not exactly, what I wanted, but it's also an option.
2. I'll investigate the GAFOR topic in more detail to find a source for the data.

lordfolken, 2021-12-27:
The source of the file i posted it seems to be here: https://www.ulforum.de/ultraleicht/forum/1_allgemeines-und-aktuelles/6003_platzrunden-datei-incl-ul-platzrunden

https://mapgen-data.sigkill.ch/manifest this is the current manifest for maps. Additional layers could be added there. 

Although noise sensitive regions etc, would be probably better realized as airspace files. that way they are included in the routing and warning logic.

volkmarnissen, 2022-01-16:
After some investigations, I found the circuits and I found out how xcm files can be changed.
I added the airspace file to my xcsoar instance. They are already a big step ahead.

If you use the airspace files, the circuits will be displayed independant of the scale. This is a disadvantage.
It would be better to see them only in zoomed scales.

For noise sensitive regions, it is important to follow the circuits as good as possible. So, it's important to see them in the map in large scales.

For the GAfOR topic, I found out, that there is an API available at dwd. It deliveres more than enough data. It's designed for internet service provider and will be charged.
This is an option. A good skydemon  integration might also be an option.

I will first concentrate on the circuit topic and try to integrate the circuit file into an xcm file.

lordfolken, 2022-01-17:
i had a look at enroute, an  opensource navigation app (android only afaik) for flying. They have better suited maps for your purpose including the approach pattern. They have weather at least metar. What i wasn't able to see is the noise sectors.

volkmarnissen, 2022-01-17:
Enroute looks promising
Only radio control and GAFOR weather info is missing.
May be it's good to use two different apps for motor and for soaring.
I will try this approach first.

Noise sectors shouldn't harm if you follow the official approach.

closed 2022-01-17 by volkmarnissen

lordfolken, 2022-01-17:
#833 
I'm have added support (in that PR) for more waypoint types. Also including reporting points, VOR,NDB...