XCSoar/XCSoar #816: Waypoints

reuti1697, 2021-12-27:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

reuti1697, 2021-12-27:
[WP-Reuti-Test.txt](https://github.com/XCSoar/XCSoar/files/7780947/WP-Reuti-Test.txt)

MaxKellermann, 2021-12-28:
You did not specify the latest version which does not have the problem.

reuti1697, 2021-12-30:
I'm sorry, I can't quite tell. I always use the latest version of Google Play.
I only noticed the error last week when I was creating new waypoint files. At first I thought there was something wrong with the waypoint files. But they seem correct to me. Thanks for your help and especially thanks for the great program. I love it. Can I do something for you? Peter.

reuti1697, 2021-12-30:
[Test.pdf](https://github.com/XCSoar/XCSoar/files/7792367/Test.pdf)

lordfolken, 2021-12-30:
So i tested several versions and it seems to be connected to the search short name search feature, introduced in v7.19
https://github.com/XCSoar/XCSoar/commit/6099befc0be5ce821b792e0ed61c1fc8bfa2ec72

* This happens with cup files that have field 2 (shortname) defined. 
* The particular file here has only few waypoints, so all the waypoints appear without searching.
* You can also trigger it when you use another filter criteria like direction, waypoints with shortname defined appear twice. 
* We knew it happens when 

Some potential approaches:
  * Seperate the name / shortname searches and use different fields in the "Select Waypoiint" Dialog
  * Find a way to eliminate duplicates in the radix tree. 
  *

MaxKellermann, 2021-12-30:
See https://github.com/XCSoar/XCSoar/pull/743

scls19fr, 2022-05-07:
I'm also facing same issue when selecting type Landable for example

[image]
![Screenshot_2022-05-07-08-16-59-30_943b36a15770a48ab82837a5cf06e5e4]

But not when filtering by distance

(https://user-images.githubusercontent.com/109167/167241580-2afc3909-e974-4d09-a1d3-3ddebdc5dda5.jpg)

scls19fr, 2022-05-07:
I don't know if that's related but removing waypoints file(s) in config and waypoints are still present on the map as if XCSoar was silently loading waypoints whereas I explicitely ask for no waypoints (or there is a kind of cache). Maybe it could explain why they are displayed twice.

scls19fr, 2022-05-07:
[image]

[image]

yo7ckp, 2022-07-05:
I gave up the Waypoint Code

commit 98a76d31 (2022-08-22): Eliminate duplicate waypoints in list

Retains filtering name and shortname but deletes duplicates
(e.g. when the shortname is the same as the name).  Also
eliminates duplicates when distance is the filter.  Bonus: this
should slightly speed up rendering of the waypoints on the map.
Closes #958
Closes #816
PR #983 (2022-08-22): Eliminate duplicate waypoints in list
Retains filtering name and shortname but deletes duplicates
(e.g. when the shortname is the same as the name).  Also
eliminates duplicates when distance is the filter.  Bonus: this
should slightly speed up rendering of the waypoints on the map.
Closes #958
Closes #816

Brief summary of the changes
----------------------------

Related issues and discussions
------------------------------
closed 2022-08-22 by MaxKellermann