XCSoar/XCSoar #817: cannot load

pby51, 2021-12-28:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

pby51, 2021-12-28:
I cannot download the map from Ireland high res. By restarting XCsoar I get a error code "No center(262)'DC 0.05'".
my phone is a samsung A12  Android version 11 last updat 1 November 2021.

closed 2021-12-28 by pby51

lordfolken, 2021-12-28:
The "No center" is a problem with the airspace file. 
The 404 for the IRE (and all other airspaces i fixed with the recent commits in https://github.com/XCSoar/xcsoar-data-content
thx for reporting.

pby51, 2021-12-30:
Are you saying that there are corrections available?
How can I get them in the xcsoar 7?

lordfolken, 2021-12-30:
I don't know what airspace file you are using. 
The center message is an error in the airspace file and has no relation to the map.

pby51, 2021-12-30:
Ok, but I cannot download the IRE map, how can I make this work

Outlook voor iOS<https://aka.ms/o0ukef> downloaden

lordfolken, 2021-12-30:
Double tap on screen->config-config->file manager->add->ire_highres.xcm then go to config->system -> site files map database ire... 

We have a general quickstart guide here
https://xcsoar.org/discover/manual.html

pby51, 2021-12-30:
Thanks it works

Outlook voor iOS<https://aka.ms/o0ukef> downloaden

https://xcsoar.org/discover/manual.html<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fxcsoar.org%2Fdiscover%2Fmanual.html&data=04%7C01%7C%7Ca73943e99e3e4138860408d9cbb07088%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637764780920571335%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=MUCxRQjbJCcMBadnuXq1nSoZiTYhkkeeSxVB5H0Gp5A%3D&reserved=0>