XCSoar/XCSoar #818: nonsensical final height calculation result

hjr, 2021-12-30:
XCSoar versions having and not having the problem
-------------------------------------------------
Observed on 7.21 (Android).
 
Expected behavior
-----------------
On final flight, checking on arrival height with different MC settings. Ca. 15km/h headwind, 30km out to fly. XCV connected and used to change the MC settings. Arrival height on display for MC=0.0 and 0.2 was reasonable and about 30-50m different (let's say 600m). 
Expected for MC=0.1 about a similar result.

Actual behavior
---------------
For MC=0.1 the result was surprisingly -26345m, something very unreasonable.

Steps to reproduce the behavior
-------------------------------
not sure.

Do you have any idea what may have caused this?
-----------------------------------------------
no

Do you have an idea how to solve the issue?
-------------------------------------------
first try to reproduce.

hjr, 2022-01-04:
https://user-images.githubusercontent.com/415029/148131898-374ad1ae-83e2-4fef-8823-28a9b116f651.mp4

Just another observation that fits into this topic. Some data might have not been available for a moment and EA height was a lot below path, and then the next moment above.

MaxKellermann, 2022-01-07:
I don't understand what the video is about.
About the final height calculation: I tried back and forth for some time, but could never see such bad results. Did you ever reproduce this?
I fear we can't do much until we have more hints on how to make reproducing it more likely.

hjr, 2022-01-09:
I am a bit clueless on this too. Will continue to try to reproduce. For now I just wanted to collect things here.

iltis42, 2022-03-30:
I have seen something like in the video in the mountains as well. What may have happend is that XCSoar calculated a longer route as terrain might have been in the way (final glide through terrain), and at one point since there was a bit lift, this wasn't needed anymore. This might also depend on the Mode of the Routingplan (Routenplaner dt,) and Reachability (Reichweite dt.) settings.

hjr, 2022-03-30:
It might depend on the routing mode, yes. I do use the "Turning" reach mode. But be aware, the issue differs slightly to what I showed with the video. I could guess there is a similar root cause.

MaxKellermann, 2023-05-09:
Apparently, this has never been observed again.

closed 2023-05-09 by MaxKellermann