XCSoar/XCSoar #819: Debug option on NMEA port is applied to the wrong port

lordfolken, 2022-01-01:
XCSoar versions having and not having the problem
-------------------------------------------------

Has the problem: V7.21

Expected behavior
-----------------

Going into the Devices dialog and pressing "debug" on a port, should yield a log.nmea file of that port.

Actual behavior
---------------

When hitting debug on port "B" the resulting nmea file actually contains the data of port "A".

Steps to reproduce the behavior
-------------------------------

Open Device Configuration
Configure a separate device on port "B" and hit the Debug option.

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

iltis42, 2022-01-03:
No i understand why i have never seen there the data i expected :)  Looks like some bug that is in since long...

MaxKellermann, 2022-01-07:
> Going into the Devices dialog and pressing "debug" on a port, should yield a log.nmea file of that port.

No, this is not expected behavior. This is not the NMEA logger, that's a different feature/setting.

"Debug" dumps all explicit protocol communication with the device, which is everything *except* NMEA. For example, it dumps data sent and received for task declaration, for submitting MacCready, flight download etc.

closed 2022-01-07 by MaxKellermann

iltis42, 2022-01-07:
Regarding:  file actually contains the data of port "A".   
@lordfolken  What is meant with file here, the xcsoar.log ?