XCSoar/XCSoar #820: MC sync to device (XCV) not working any more

hjr, 2022-01-01:
XCSoar versions having and not having the problem
-------------------------------------------------
7.21 introduced the issue, 7.20 was fine.

Expected behavior
-----------------
Change MC value on XCS, see the value propagating to a connected device like XCVario. Driver sync to device feature switched on.

Actual behavior
---------------
XCS remains silent. XCVario NMEA monitor does not show the reception of the !g sentence. 

Steps to reproduce the behavior
-------------------------------
Eg. PC with XCS and connected or simulated XCV device, sync options enabled. Change MC on XCS, monitor NMEA data stream. 

Do you have any idea what may have caused this?
-----------------------------------------------
not really

Do you have an idea how to solve the issue?
-------------------------------------------
neither

iltis42, 2022-01-01:
The events NMEA string sending mechanism in .xpi files using "sendNMEA1|2" method doesn't work either in 7.21. Search for that took me several hours, but is confirmed by another user on phone (who searched several day's). Do we consider this as same area, or better a second issue ?

MaxKellermann, 2022-01-07:
I cannot reproduce this with a TCP port (and socat).

MaxKellermann, 2022-01-07:
@iltis42, no different issue.

hjr, 2022-01-07:
could it be just a build error of the 7.21 apk target?

iltis42, 2022-01-07:
> could it be just a build error of the 7.21 apk target?

Confirmed on Android target with the 7.21 Version from xcsoar download page.

hjr, 2022-01-16:
Ok, the issue was introduced on XCV not accepting NMEA messages that are terminated with just one single "\r". Or better I dont know if XCS changed on its NMEA termination. Nevertheless it is resolved by XCV accepting a single \r NMEA termination.

closed 2022-01-16 by hjr