XCSoar/XCSoar #821: Wind arrow overlay contrast/visibility on map is low, needs improvement

hjr, 2022-01-01:
XCSoar version
--------------
Head, any.

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
[
[image]
](url)
As this screenshot demonstrates a bit: The wind arrow sometimes is hard to recognize due to the background it overlays. A white contour like for glider icon and wind speed would do better.
Since the wind arrow always points towards the glider icon redndering the well visible wind speed number in the middle of the arrow would  probably do the trick already.  

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
Always be able to recognize the current wind with just a glimpse. Yes, in Worcester the Wind changes after every 100meter climb and around every next corner of the ridge. XCSoar wind calculation does quiet well, if you just would be able to pick it up quickly from the screen.

lordfolken, 2022-01-01:
For the moment, there is the wind infobox. There should be no contrast issue there.

lordfolken, 2022-01-01:
In general all map elements should be encased in a white outline. That way it would be readable in any situation.

MaxKellermann, 2022-01-05:
It already has an outline, albeit a black one. And it has a fill, but that's translucent on OpenGL, so it doesn't obstruct the map too much.
One could argue that it doesn't obstruct the map enough, but then others will say it's too prominent and distracting.

lordfolken, 2022-01-05:
The correct way to do it is to have a white and black outline. Naviter is doing it right:
[image]

MaxKellermann, 2022-01-06:
I've moved the speed label:
[image]
But maybe having a "black symbol with white outline" should really be a good standard for all map overlays that are not geo-referenced? (If it doesn't have a white outline, it's an item on the actual map).

hjr, 2022-01-07:
All my buddies flying with challenging wind conditions in Worcester SA right now do appreciate, thx!

closed 2022-01-09 by hjr