XCSoar/XCSoar #822: Issues in Flarm IGC download procedure

speed-axel, 2022-01-01:
XCSoar versions having and not having the problem
-------------------------------------------------

Which XCSoar version are you using when seeing the problem? Android version 7.20 from the play store.
Which XCSoar version is the last one you know that did not show the problem? Never used xcsoar before

Expected behavior
-----------------
Flarm IGC download protocol procedure has to follow.

Actual behavior
---------------
When the flarm flight list is opened, the command $PFLAX is sent to the flarm to switch it in binary mode. After that a first binary ping command is sent to the flarm as:
73 08 00 **00** 01 00 01 53 2a
It has the following content:
 1. byte start frame
 2. and 3. byte length of data frame
 4. byte version
 5. and 6. byte SeqNo
 7. byte message type
 8. and 9. byte CRC over message
 10. Payload (MSG type and more)
 The general issue is the setting of the version byte. According to the Flarm Bincom spec it should be set to 1 or higher.
A Classic Flarm will never answer to the first ping. Therefore after a timeout a second ping command is sent to the flarm device. But this ping command is always one byte to long:
73 08 00 00 02 00 01 03 78 **31**
The flarm device seems to be able to handle that but that is a bug in xcsoar.

Steps to reproduce the behavior
-------------------------------
Trace the data stream, when the Flarm flight list is opened.

Do you have any idea what may have caused this?
-----------------------------------------------
program issue, maybe the send buffer is not cleared before filling?

Do you have an idea how to solve the issue?
-------------------------------------------
fix it in the program code

iltis42, 2022-01-01:
I can see exactly the same issue using 7.20 Android plus an older classic flarm version 7.0.3.
Below trace (1) is what i receive in the bluetooth from xcsoar and forwared to serial interface S1 in my BT to serial layer in xcvario at that time when the second binary Ping is received.

The code itself where the BinaryPing is created is here, at a first glance i can't see anything odd: src/Device/Driver/FLARM/Mode.cpp

When trying to reproduce this on Linux i must see that the whole binary download apparently is broken at the moment in Android 7.21 and also in latest Linux pulls. Guess this is worth a second issue here.

1) Trace
I (00:01:26.231) BTSender.cpp: data avail for BT send 13 bytes: $PFLAX,A*2E
I (00:01:26.618) BTSender.cpp: BT RFCOMM RX
I (00:01:26.619) BTSender.cpp: BT data received 10 bytes
I (00:01:26.619) BTSender.cpp: 0x3ffe3c74   73 08 00 00 02 00 01 03  78 31                 |s.......x1|
I (00:01:26.633) Serial.cpp: S1: TX len: 11 bytes
I (00:01:26.637) Serial.cpp: 0x3ffe477c   73 08 00 00 02 00 01 03  78 31                 |s.......x1|
I (00:01:27.126) BTSender.cpp: BT RFCOMM RX
I (00:01:27.126) BTSender.cpp: BT data received 9 bytes
I (00:01:27.127) BTSender.cpp: 0x3ffe3c74   73 08 00 00 03 00 01 33  44                    |s......3D|
I (00:01:27.140) Serial.cpp: S1: TX len: 10 bytes
I (00:01:27.145) Serial.cpp: 0x3ffe477c   73 08 00 00 03 00 01 33  44                    |s......3D|

lordfolken, 2022-01-01:
https://github.com/August2111/meta-openvario/commit/a28385e1d997de52dce86065c27c5bf11c82c51d potenially?

iltis42, 2022-01-01:
Hmm, should be an older thing either as its in 7.20 Android in PlayStore that is older than 29th of November.

The surplus 0x31 is likely the ESCAPE_START = 0x31, and accidently appended, btw. START_FRAME = 0x73.

Now we needed to understand, how can happen...

 // Check for bytes that need to be escaped and send
    // the appropriate replacements
    if (*p == START_FRAME) {
      port.Write(ESCAPE);
      port.Write(ESCAPE_START);
    } else if (*p == ESCAPE) {

speed-axel, 2022-01-02:
I observed an other issue. With Flarm release 7.01 the binary switch command $PFLAX is acknowledged by Flarm. Xcsoar sends the $PFLAX command together with the first binary ping command as to wait for the Flarm response.

The Flarm BINARY COMMUNICATION Specification writes the following:

_1.1.1
Description
Initiating Binary Mode (BP)
Switching to BP is initiated through TP. Connection should first be checked by sending a ping sequence in TP. Binary mode is engaged by sending the text command $PFLAX including <CR><LF> (0x0D0A) to FLARM. If the command is successful, the device will reply $PFLAX,A through TP before switching to binary mode. In case of error, $PFLAX,A,ERROR,<error> will be replied through TP and the device will remain in TP. The command $PFLAX is not accessible during flight; $PFLAX,A,ERROR,INFLIGHT will be received in that case.
**After switching, the connection can be checked by issuing a ping through BP.**_

You should require the document FTD-026 from Flarm where the binary protocol is described.
Furthermore consider the fact, that a PowerFlarm answers to the first binary ping command and that it switches alone to the text mode, if 60s nothing has happened in binary mode.

r0w4n, 2022-02-27:
Is this the same issue that I'm experiencing that I captured in this [screen cast](https://youtu.be/MfnkRDKrXMU)?

closed 2022-05-26 by lordfolken