XCSoar/XCSoar #824: XCVario to XCSoar ballast synchronisation errors

GrahamD2, 2022-01-03:
Hardware 
XCvario firmware v21.1230-17
Huawei P30 lite Android v10.0.0.491 XCSoar v7.21
Bi-directional data exchange over BT using XCV drivers and protocol
Earlier versions of XCS exhibit problem.

Problem
This may relate to issue #807
Loaded into XCV, dry mass, pilot weight, and polar reference mass expressed as wing loading kg/m2
and in XCS Planes, Wing Area, Ref Mass and Dry mass (which is to include pilot according to manual).
With zero litres of ballast on both XCV and XCS when I open the flight setup page on XCS there is 42L that has been loaded,  Closing the flight setup page writes back 42l to XCV, opening the flight page again 81l is then loaded to XCS which is then written back to XCV and so this continues each time I open the flight setup page,116l then 148l and so on. Pressing the Dump button in XCS produces some very interesting results!!!
Changing ballast in XCS to zero or any other number, writes back that number to XCV but the addition process starts again when XCS Flight Setup is opened.
It is not possible to zero or synchronise the ballast between XCS and XCV when bi-directional data exchange is running. The only solution is disable the bi directional data exchange "Sync from device" with the loss of the QNH,MC and Bugs sync, 
Adjusting the dry mass and the pilot weight to equal the ref mass in XCV also overcomes the problem but at the expense of STF on the XCV.

My Observation for what its worth!
Looking at the Cambridge protocol which I think is used, in part, in the Xcvario protocol, calls for a ballast ratio made up of the total mass of the glider (dry mass, pilot, water) divided by the polar ref mass). In the case of my setup this is 1.27 with no ballast (water).
Monitoring the data stream from XCV I see the ballast variable is indeed  1.27. 
XCS seems to be treating any ratio over 1 as additional ballast. 
If I set the total mass equal to the ref mass in XCV, as suggested above, I get a ballast ratio of one on the data stream and the problem goes away, STF on XCV not withstanding.

iltis42, 2022-01-03:
Confirmed the problem on latest Android XCSoar v7.21 from downloads page: 
For my glider with max ballast 160 l if we enter 70 liter, i expect the ballast message to created like "!g,b4.375" (10*70/160 = 4.375 ). Hence we see just "!b4.3**00**" in the data monitor. The missing 0.075 of fraction of max ballast corresponds to the lack of ballast you see in the vario. The reason for that is unknown at the moment, looks like a bug in Android version of xcsoar limiting precision roughly to about ~1%, after second digit behind comma data is lost / zeroed.

No issue with latest pull built for Linux, a new Android build would be helpful here in order to rule out a persisting code issue. Several other bugs have been reported for 7.21 Android, e.g. MC sync and sendNMEA1 events are apparently broken.

loganstk, 2022-01-03:
I've been having the same issue with my LXNAV S100 this season. Given the same dry mass XCSoar always tends to display 50 kg of ballast more than the vario (even to the point where it exceeds the glider's MTOW limitation). I just checked and the issue is still here (XCSoar v7.20-Android).

Here are the values I'm using: 
[image]
[image]

If I understand correctly, the Dry Mass in XCSoar should be the same as Empty Weight + Pilot Weight in S100.

iltis42, 2022-01-03:
LX and xcsoar may differ a bit here here.  An 'Empty weight' doesn't exist in xcsoar, instead there is "Dry Mass", what includes the Pilot (as pilot not configurable). LX also exchanges just 'overload' with xcsoar.  Unclear is also what "Minimum.glider.weight" exactly means, not explained above. But somewhere here the fish might be in the butter. 
Overload is xcsoar from the dialog above is calculated as  plane.dry_mass + fraction * plane.max_ballast) / plane.dry_mass, where fraction*plane.max_ballast is your water ballast. 
So if plane.dry_mass is a different thing that minimum.glider.weight, i did not wonder that result is different.
Is there any other reference in manual what exactly minimum.glider.weight means ?

hjr, 2022-01-03:
Yip, a big issue of this protocol definition. Exchanging factors but not the norm to it, or just have a hidden understanding of what the norm (or unit) is.
Issues by design ;)

loganstk, 2022-01-03:
@iltis42 thanks for clarifying the formula used in XCSoar. Yes, I agree, that my problem is vendor-specific and probably deserves a separate issue opened. I just wanted to let people know that @GrahamD2 is not the only one facing this kind of problem with the ballast synchronization. :) Another pilot from my club also experienced similar issues with his ERA 80.

P.S. I have written a letter to LXNAV asking for some details about parameters they use for their calculations (namely the Minimum.glider.weight). I think I'll open another issue for LXNAV devices once I collect so more data.

iltis42, 2022-01-03:
@loganstk many thanks, this info's should help to understand the problem faced with LX ballast exchange.
For the issue with XCVario, i guess checking with next Android version, recent Linux version does not show this issue once reference data is same, will bring more clarity.

lordfolken, 2022-05-26:
@GrahamD2 can you try with the current combination hw/sw?

GrahamD2, 2022-06-03:
Sorry to late response.
Issue unchanged with XCS v7.23 and XCV v22.0128-16
Setting in XCV, Ref wing loading =33.19 kg/m (377kg/11.36m2) Pilot weight 80kgs. Empty glider= 343kgs.
Ballast %age = ((empty glider +pilot wt +ballast)/ ref weight) ((343 + 80+ 0 )/377)= 1.122 see screen shot from XCS monitor
Settings in XCS, Plane polar Ref Mass=377 kg, Empty mass 345kg. Flight setup Crew = 80kg
[image]

iltis42, 2022-08-22:
After some discussions, we found a solution for this, we need a warrant to synchronize reference- crew- and ballast-mass between the devices instead of exchanging just ballast overweight what will be erratic in case one of this weights is changed, leading to the observed loop. The issue can be fixed for XCVario, for Borgelt and Cambridge, those devices/protocol's are no more under development, this parameters needed to be manually be syncronized between the XCV and XCS.
The changes are described here:
https://github.com/iltis42/XCVario/issues/201
https://github.com/iltis42/XCVario/issues/202
For XCSoar we need to: 

1. Takeover of reference-, crew- and ballast-weights from new protocol messages in XCVario Protocol
2. Remove ballast overload takeover from $PXCV field (obsolete after 1.)
3. Send NMEA messages in new protocol format for all three weights from XCS to XCV

iltis42, 2022-09-14:
PR https://github.com/XCSoar/XCSoar/pull/1043  is dedicated to fix the issue

commit 56bd910e (2022-09-19): Device/XCVario: Add protocol for absolute ballast exchange

The current way sent ballast as fraction of max ballast to the device plus
another fraction (ballast overweight) from the device leads to issue #824.
The issue can be fixed sending an absolute value in litres to the device,
addressed with the new protocol version 2. In order to remain backward
compatible, XCSoar and XCVario will negotiate furthermore the protocol version.
This is initiated by XCSoar what will send its version as soon the link is
up and $PXCV messages are getting in. The additional $PXCV is needed
as an application lay [... cut]
closed 2022-09-15 by MaxKellermann