XCSoar/XCSoar #828: https://forum.xcsoar.org/ does not send activation emails after registration

JoeggiCH, 2022-01-06:
2-3 days ago I registered on https://forum.xcsoar.org/ . After registration I expected the activation email to confirm my email address.  After more than 24 hours without an email I asked for the activation email again; waited 2-3 hours and then wrote to the board administrator - but did not get a reply either.

What can I do to complete the registration ?

BR
Joerg

lordfolken, 2022-01-06:
Ok i had a look... 

The major email provider (outlook, web.de, gmail,gmx.de) are refusing mails from us on a reputation basis. 

- [x]  i changed the ehlo name to the amazon ec2 instance rdns record

- [x]  added myself as postmaster

- [ ]  hope to add @Exadios as forum admin

- [ ]  willl add an spf record to the domain xcsoar.org ```v=spf1 a mx -all```

ChristopheSeg, 2022-01-07:
Hi 
Additionally, I saw that the forum page https://forum.xcsoar.org/ is not reachable when I tried to register. (error 403 You don't have permission to access this resource.). May be this is because the forum is under maintenance. 

Christophe, new user of XCSoar

lordfolken, 2022-01-10:
I'm currently relaying the mail via my mail server. This should lessen the problems.

lordfolken, 2022-02-24:
I got no reaction from the domain owners of xcsoar.org. Hence my workaround will stay in place.

closed 2022-02-24 by lordfolken