XCSoar/XCSoar #831: Draw Point of no return / Turnaround Point on map

DavisChappins, 2022-01-11:
XCSoar version
--------------

7.21 (feature is independent of version)

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Place a "Point of no return" marker on aircraft track line where the current track line intercepts the MC0 altitude of your selected GoTo waypoint. Suggest a sideways ) mark or something similar. Small and unobtrusive.
Only applicable if your GoTo waypoint is a landable field (the automatically assigned takeoff field or another selected GoTo waypoint)
Not applicable when flying a task

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

The glide range line tells the pilot how far they can glide from their current position to that spot on the terrain. What if they want to ensure there is enough altitude to return to a known field (the selected GoTo waypoint)? This feature would put a marker at the "point of no return" or "turnaround point" to allow the pilot to know upfront how far they can go before they must turn around and still reach their selected airfield.

Turning around at MC0 = 0ft/m obviously leaves very little margin so possibly there could be a "point of no return MC value" that is selected or possibly it could use the currently selected MC or Safety MC value already in use.

lordfolken, 2022-01-11:
Interesting idea. I used the Altn GR 1 glide ratio  Infobox for this, to estimate how reachable the landable is. 
Or the final glide bar when not running on a task. 

Drawing the area from the current position on the area which you can get to and still make it your landable would make it certanly more visible for strategic planning.

lordfolken, 2022-01-24:
https://github.com/XCSoar/XCSoar/pull/778 there was a pr that wanted to adchive a similar goal. But I think this idea would be a better approach for safety.

DavisChappins, 2022-01-31:
To keep this discussion going, Commercial aircraft place a little arc to indicate the point on their lateral path where the aircraft reaches the altitude set in the mode control panel.
[image]
This function for XCSoar could be similar but a smaller character as the map has more clutter in general.

groundhogxc, 2022-08-17:
I like it, too.
But it should not only apply to landable points but to any point with altitude information, I see no reason why not.
(You might set to goto point to a specific reporting point for approach, to a triangle closing point, a mountain peak, or any other important goal)

DavisChappins, 2022-08-17:
Here is an image of how it should look, a little black arc on the track line indicating the point in which the final glide bar would equal 0 for the selected MC setting. This will tell you how far you may go before turning back towards your GOTO point. The GOTO can be a landable airfield or any waypoint.

I'd like to call it the PNR Arc (point of no return arc)

This would be helpful for two main reasons (both I wish I had on a recent flight):

1) Can I make it somewhere before I have to turn around? Maybe a cloud or a mountain ridge. Yes my glide amoeba tells me I can make it _to there_ but once I get there _CAN I MAKE IT BACK?_ If the PNR arc is at or past the mountain ridge or cloud then yes, I can safely make it back if there is no lift at the ridge or cloud. But, if the PNR arc is before the ridge, I am committed to another landing point because my current GOTO waypoint will be out of reach.

2) How far can I overfly on final glide to maximize OLC score? I plan final glides with plenty of margin and overshoot the airfield by 5-20 miles, depending on altitude, then make a u turn and fly back and land. If I encounter sink along the way then I am still safe. When flying past the airfield I feel the air and expect to fly through the same air on the way back so I get an idea of how accurate my arrival altitude will be.
Using the PNR arc I can get a visual picture of how far I can go before turning around to land and better plan out my route.

Unfortunately I can not code C++ so I hope someone will pick this up and implement it. I am willing to contribute to the documentation defining this feature with images, descriptions, and use cases.

See the image below, once I cross the arc the final glide bar will be at 0 and I will need to turn around or pick another place to land.
[image]

closed 2026-09-10 by WolfgangDrescher