XCSoar/XCSoar #832: Draw other glider's thermals on map as Thermal markers - from FLARM data

DavisChappins, 2022-01-11:
XCSoar version
--------------

7.21 (feature is independent of version)

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

XCSoar currently leaves a thermal marker when you climb, so you can refer to it later. I believe if you tap it, it shows strength, time since, and maybe something else.
This functionality could be expanded to FLARM targets as well. If a FLARM aircraft stays relatively stationary and shows a 30s climb avg of more than some value, place a marker on the map similar to the thermal marker (different color maybe) with that climb rate, time since, and maybe other data when tapped on.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

This functionality would further enhance the thermal awareness of the pilot by tracking and recording other glider's thermal locations and strengths. A pilot could receive the same information if they focused on the map watching every glider, their position, climb rate, etc instead of flying. This would automate that task and leave little thermal marker breadcrumbs for the pilot to fly towards.
I do not believe LX or Oudie has this feature but it would be very helpful in recreational flying as well as competitions.

lordfolken, 2022-01-11:
Yes i thought of this aswell. 
In general seeing tracks from other gliders colored according to their sink/climb rates would be interesting. 

Combine this with an ogn client, and you have a very large range and see where what is happening. 

Currently we have:
The https://cloud.xcsoar.org feature will share thermals amongst xcsoar users.
There is the beginnings of a Therminfo support
There was a now dormant pull request for skysight overlays.

kobedegeest, 2026-08-28:
If people have good flarm and NMEA data with lots of traffic that they are willing to share that could be useful for testing this, my own data is only from ogn viewer linked to xcsoar