XCSoar/XCSoar #834: Implementing RC Electronics driver for Fenix unit

andrejvrecer, 2022-01-12:
XCSoar version 7.21
--------------

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
New driver support for communication with Fenix unit from RC Electronics.
Currently unit already supports data output of GPS NMEA data (GPGGA, GPRMC, GPRMB), Flarm data  ( if Flarm unit is conencted to it) and LXWPx data

In specification which can be found on website: https://www.rc-electronics.eu/downloads-gliding/manuals-gliding/ There is RCDT (data transfers) needed to be implemented so different data can be transferred from XCSoar to Fenix unit and vice versa if needed. Specially TASK declaration with zone shapes and pilot information

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

lordfolken, 2022-01-13:
Are you the manufacturer? Ideally the manufacturer implements **and  maintains** the driver in XCSoar. That way  compatibility with for example firmware upgrades, is ensured.

lordfolken, 2022-02-24:
No reaction. Closing this issue.

closed 2022-02-24 by lordfolken