XCSoar/XCSoar #835: enable rotary encoder for OpenVario not depending on HasPointer

Blaubart, 2022-01-13:
With OpenVario, the query HasPointer makes little sense when using the arrow keys, because it asks whether the library libinput is installed. If so, HasPointer is true and the arrow keys work as if a pointer device were installed. The majority of users use the Openvario without a touchscreen and still have the library installed in the image. From my point of view, the query therefore makes no sense with the OpenVario. It would therefore make sense if there was another query that switches HasPointer off for the function of the arrow keys in the OpenVario. Since a patch is not a good solution, could such a query be built into XCSoar?

https://github.com/Openvario/meta-openvario/pull/170

MaxKellermann, 2022-01-13:
> because it asks whether the library libinput is installed.

Where does it do that?

Blaubart, 2022-01-13:
https://github.com/XCSoar/XCSoar/issues/625#issuecomment-853140311

MaxKellermann, 2022-01-13:
There I don't see a check for whether libinput is installed.
Maybe you misunderstood this one:
>  The `HasPointer()` check is only implemented for XCSoar builds targetting the Linux console with libinput.

... which is the opposite of what you assumed. If libinput is **not** used, then XCSoar assumes there is a pointer device.
If libinput **is used** (e.g. OpenVario), XCSoar asks libinput whether a pointing device is present.

Blaubart, 2022-01-13:
OK, I got that wrong indeed. But it appears that XCSoar assumes that a pointing device is being used, even if no touchscreen is installed. But there are pilots who use the stick remote control and the touchscreen. They mostly use the touchscreen on the ground, so that the full functionality of the arrow keypad would be useful in flight. So, from my point of view, it would make sense to switch off this distinction in the OpenVario.

MaxKellermann, 2022-01-13:
> OK, I got that wrong indeed. But it appears that XCSoar assumes that a pointing device is being used, even if no touchscreen is installed.

XCSoar asks libinput, so what does libinput say? Which device is declared to be a pointing device by libinput?

Blaubart, 2022-01-13:
no idea what libinput is saying. Obviously, a touchscreen is there as a pointing device, even when there is none. This is then obviously declared as a pointing device even if it is not available. But apart from that, with the OV it doesn't always make sense to castrate the arrow keys despite the presence of a touchscreen.

lordfolken, 2022-01-13:
In case of the remote stick for openvario: 

The stick has both keymode and mouse mode. So you can activate the mouse mode by long pressing a key and then you can pan the cursor around with the directional arrows. 

So the stick emulates usb mouse and keyboard at any given time. Potentially that could have an impact on the above.

Blaubart, 2022-01-13:
That is known. But do you really want to stare at the screen while flying while you're desperately trying to hit the field with the "mouse" when you could simply use the left and right to go through the fields one after the other??? It's really not functional.

lordfolken, 2022-01-13:
The point is: The HasPointer() function returns false, despite there being pointers present. Which is why the Previous/Next is being activated even though its not intended by the design to be active if there are pointers present.

Currenlty its less this part of the code, its more that the ifdef in the build systems which are probably wrong.

Blaubart, 2022-01-13:
I have to say, I'm not a programmer, but my understanding is that the HasPointer() function is true even though there is NO pointer (touch screen). For this reason, Previous/Next is deactivated, which is why the right and left can no longer be used, for example, to control the submenu for the respective device in the case of NMEA devices. But it may well be that I misunderstood!

MaxKellermann, 2022-01-13:
The list control is hard-coded, but this isn't about the list control, but about dialog-global hot keys for focusing buttons; these are intercepted before they reach the list control.

closed 2022-01-14 by MaxKellermann

Blaubart, 2022-01-14:
Thanks a lot!!!!

MaxKellermann, 2022-01-14:
> I have to say, I'm not a programmer, but my understanding is that the HasPointer() function is true even though there is NO pointer (touch screen).

And.... guess why that is!
https://github.com/Openvario/meta-openvario/blob/a407dccd8541dd715a165dff64e50b2969e51e80/recipes-apps/xcsoar/files/0007-Disable-touch-screen-auto-detection.patch#L24

OpenVario patches XCSoar and forces it to always assume there's a touch screen.
@Blaubart, @kedder

kedder, 2022-01-14:
> OpenVario patches XCSoar and forces it to always assume there's a touch screen.

Yeah, that makes buttons bigger. To my defense, there are openvarios with the real touchscreen too! :)

MaxKellermann, 2022-01-14:
If you want to make buttons bigger, patch `GetMinimumControlHeight()`, but don't force it to have a touch screen... that has surprising side effects.

kedder, 2022-01-14:
I'll try to update the patch.