XCSoar/XCSoar #837: Skysight wheater

Pikglider, 2022-01-15:
It would be a nice feature, to get the skysight wheater as overlay for the moving map. Especially the wave parameters.

lordfolken, 2022-01-15:
There are sadly several efforts that have not meet contribution qualtiy. See:
* https://github.com/XCSoar/XCSoar/pull/461
* https://github.com/XCSoar/XCSoar/pull/298
Maybe care to revive them?

hjr, 2025-02-09:
Could we merge (take somehow / integrate) the effort from Jetsada? I like the way weather info sources are put together in the configuration.

lordfolken, 2025-02-09:
Please take the time and rebase the code, and do a PR draft. We can work on it.

dssherrill, 2025-06-04:
I can't think of a reason to remove TCHAR.  It is a simple one-line macro that is easily redefined to the right thing on each platform, expanding to either char or wchar_t.

   ...david

DanD222, 2025-06-04:
Looking at latest the OpenSoar Skysight offerings, I'm not sure if XCSoar should aspire to take this over from OpenSoar.

[image]

[image]

August2111, 2025-06-14:
Well, what can I say on that topic... Of course, I took advantage of Caz's great preparatory work for my SkySight component! I didn't really understand some things the first time, why he did things the way he did them. I had to change some things to suit my own needs so that they would be clearer for me and perhaps for others (e.g., renaming 'metric' to 'layer' - as it's called in SkySight). I had to completely change some things to make it run error-free and error-tolerant. However, I also summarized some changes right away - otherwise I wouldn't be finished next year... I also noticed a whole series of things that weren't yet suitable for a reliable and stable connection. In that respect, I personally wouldn't be particularly happy if the wheel was reinvented and the component was rebuilt on XCSoar in a second form. I can't prevent that - it's open source, after all - but it will be very difficult for me to merge the heavily modified parallel component back into my Folk, as I do with all other components. However, I don't even claim that the SkySight component is fully developed yet, because there is (unfortunately!) still a lot to do... At the moment I am very happy that SkySight on OpenSoar seems to be running very reliably and stably: I went to the World Championships in Tabor as a helper for Stefan Langer – and so far, Stefan hasn't noticed any crashes or anything similar with OpenSoar and active SkySight (unlike my 'first attempt' at the beginning of April, which I marked as unsafe on my release page because it crashed when the internet connection was unstable!). I also can't quite follow Dan222's comment, who wants to use screenshots to show that my component is doing stupid things just because I added the *.txt extension to the temporary text and log files, which XCSoar (and OpenSoar too) unfortunately interprets as an airspace file... (The problem has long since been resolved.)
Regarding TCHAR, all I can say is: Of course, in my opinion, it's not absolutely necessary to completely convert to ASCII or UTF8 strings BEFORE integrating SkySight, but...: Internet components use CURL, which requires 'normal' text strings - so you always have to switch from TCHAR strings to char strings, i.e., you always have to know exactly why you're doing something and how ;-), which is unfortunately not possible with the simple macro. I don't know how many PRs I've seen where the developer is repeatedly unclear about this exact problem! These transformations make the component harder to understand - and potentially more prone to errors! Thirdly (see above), a comparison between OpenSoar and XCSoar is not possible at this point, or only possible with great effort.
And 4th: TCHAR and WideStrings (UNICODE) are only available on Windows; all other targets work very well with UTF-8, which makes the code more readable, easier to understand, requires no transformations, and also allows working with standard components and functions! I dare not say how many XCSoar applications run on Windows, but I can only say that it's not necessary on Windows either: 90-95% of my development work is done on Windows – and I don't need WideChar/TCHAR/UNICODE! A complete overhaul of the source code ultimately only takes a few steps, but the problem is that it involves touching and modifying many thousands of files. I've already taken this step with OpenSoar – and discovered in a new branch that this step can be performed reliably and without problems!
Please understand that I am urging XCSoar to finally address this issue as well. If I understand Folken correctly, he sees the problem in a similar way to me!
I've made a lot of other changes and improvements on OpenSoar that would certainly be a good fit for XCSoar as well – I probably have the most time of any of you to make progress on this project right now – and please understand my attitude that, especially with such an important system issue, I insist that we allow no or very few differences in the 'underpinnings'!
At this point, I would also like to emphasize again that I would very, very much like to 'return' to XCSoar, if only because the entire infrastructure will then no longer be a one-man show, which also consumes an incredible amount of my energy – but in my opinion, a compromise must be possible on both sides ;-)

lordfolken, 2025-06-16:
I generally agree that the TCHAR issue is a hassle and that we should find a different way to do this. Can we have a discussion here on how to approach this:  https://github.com/XCSoar/XCSoar/discussions/1800