XCSoar/XCSoar #838: Profile file destroyed after Android shutdown

hjr, 2022-01-15:
XCSoar versions having and not having the problem
-------------------------------------------------
Observed on 7.20 and 7.21 running on Android 9

Expected behavior
-----------------
A regular shutdown of the Android device should exit properly from a running XCS and save all profile settings. 

Actual behavior
---------------
At least the profile changes are lost. The profile file might be chopped or destroyed. Widnesed three lines long profile (very incomplete) files which have been complete and functional before. 

Steps to reproduce the behavior
-------------------------------
Start XCS on Android, add eg. a new screen with different map/info box layout. Shutdown the Android device. Restart Android and XCS, check if new screen is still there.

Do you have any idea what may have caused this?
-----------------------------------------------
The Android shutdown event is not handled properly.

Do you have an idea how to solve the issue?
-------------------------------------------
Assume some Java wrapper issue to cope with that. The X Windows exit event seem to work fine with XCS.

lordfolken, 2022-01-15:
Do you have a sample "destroyed" file?

MaxKellermann, 2022-01-24:
Did XCSoar crash? What do you see in the logcat after this corruption occurred?

hjr, 2022-01-25:
Will collect more info on this topic soon.

Best

Helmut

hjr, 2022-02-08:
Not able to collect decent evidence and reproduce. Close it for now.

closed 2022-02-08 by hjr