XCSoar/XCSoar #839: Dark text background on Debian 11 compiled KOBO version 7.21

fraktal-sb, 2022-01-15:
XCSoar versions having and not having the problem
-------------------------------------------------

Which XCSoar version are you using when seeing the problem?

v7.21 compiled at Debian 11 (see screenshots)

[image]
[image]

Which XCSoar version is the last one you know that did not show the problem?

v7.20 is OK and the KOBO version from XCSoar homepage does not have this behavior

Expected behavior
-----------------

Texts should not have a dark background

Actual behavior
---------------

Texts have a dark background; this is unreadable; make clean and recompile did not help

Steps to reproduce the behavior
-------------------------------

use standard Debian 11 and compile with "make TARGET=KOBO DEBUG=n"

Do you have any idea what may have caused this?
-----------------------------------------------

no

Do you have an idea how to solve the issue?
-------------------------------------------

no

ulipo, 2022-01-15:
Same here, compiled on Docker image, and same also with KoboRoot.tgz downloaded form official site.
[image]

MaxKellermann, 2022-01-24:
cd175cbf7d018dd44e7cb6c32b43f61e37293214 is the first bad commit

closed 2022-01-24 by MaxKellermann