XCSoar/XCSoar #845: PilotEvent should be accessible from QuickMenu

lordfolken, 2022-01-23:
The quickmenu lacks an option to activate PilotEvent. 
This is a problem, for keyboard-less devices like the openvario.

lordfolken, 2022-04-26:
@August2111 do you have a proposal?

August2111, 2022-04-26:
Yes, I have.
But I had done a lot of work with XCSoar regarding a whole range of topics and problems that I see (WeGlide upload, task download, downloading FLARM flights, serial port in Windows and, and, and) but relatively little is happening in the system at the moment, since Max has been somehow unavailable for over 10 weeks :-( - and his feedback is necessary for a whole bunch of these things...
Maybe you know when Max is coming back?

lordfolken, 2024-05-21:
This is implemented.