XCSoar/XCSoar #847: Kobo Clara HD screen sizing wrong since recent change

GeekGoneOld, 2022-01-25:
Recent changes (after release of 7.22) have Kobo Clara HD only using the upper right quadrant (i.e. 1/4 of the screen).  What's displayed is an entire screen (i.e. not cropped but scaled.

XCSoar versions having and not having the problem
-------------------------------------------------

It fails with b7890b commit and worked with the 7.22 release.

Expected behavior
-----------------

The display should use the whole screen

Actual behavior
---------------

the display only uses 1/4 of the screen

Steps to reproduce the behavior
-------------------------------

run build b7890b on a Clara HD

Do you have any idea what may have caused this?
-----------------------------------------------

likely a recent change

Do you have an idea how to solve the issue?
-------------------------------------------

best way is to look at the recent changes and see if any affect DPI or scaling (as I think they do)

MaxKellermann, 2022-01-25:
Please bisect

GeekGoneOld, 2022-01-25:
Will do so tonight or tomorrow.

GeekGoneOld, 2022-01-26:
First commit to show this behaviour is 41363c000f988df011fa8b68e75912c13877ce8a.  This was the last commit of Jan 23.

fraktal-sb, 2022-01-26:
(possibly) same on KOBO GLO HD, see screenshot
[image]

MaxKellermann, 2022-01-26:
Every bug becomes shallow after a bisect!

closed 2022-01-26 by MaxKellermann

GeekGoneOld, 2022-01-26:
I will remember the bisect.  Should have done it in the first place!

> (possibly) same on KOBO GLO HD, see screenshot [image]

@fraktal-sb  how did you do your screen dump?

fraktal-sb, 2022-01-26:
> @fraktal-sb how did you do your screen dump?

see

https://forum.xcsoar.org/viewtopic.php?f=7&t=3585&p=20263&hilit=kobo+screenshot#p20263

at linux you can do (example)

ffmpeg -f image2 -vcodec rawvideo -s 1088x3072 -pix_fmt gray -i ftp://192.168.178.20/screen.raw -vf "crop=1072:1448:0:0" /home/josef/Schreibtisch/out.png

instead of using irfanview to convert from raw to png

GeekGoneOld, 2022-01-26:
Thanks.  And thanks for the link.  I'm running an "intro to XCSoar" course
in 2 weeks and could use a few screenshots.

Keith