XCSoar/XCSoar #849: Black screen on linux

Scumi, 2022-01-27:
XCSoar versions having and not having the problem
-------------------------------------------------

last working commit:  9c7ca67a0259105e0d7e0c1beb853da9cf43bcf6
O/S: Linux Mint 20.2

Expected behavior
-----------------
XCSoar starts and shows map and infoboxes

Actual behavior
---------------
XCSoar only shows a black window

Steps to reproduce the behavior
-------------------------------
Compile on linux (UNIX target)

Do you have any idea what may have caused this?
-----------------------------------------------
Commit cde2c7261d8fcd8607f9e0ba02b67e47abc49aff

Do you have an idea how to solve the issue?
-------------------------------------------
No

MaxKellermann, 2022-01-27:
Which graphics driver? Which kernel and Mesa version?
Do you see any GL related error messages on stderr or the log file?

Scumi, 2022-01-27:
Graphics driver: nvidia
Kernel: 5.4.0-96
OpenGL version string: 4.6.0 NVIDIA 460.91.03

No differences in the logs between both commits (apart from the commit hash):
```
Starting XCSoar XCSoar v7.23-Linux~git#cde2c7261d
Display dpi=96,96
OpenGL: npot=1 stencil=0x84f9
Initialise fonts
```

MaxKellermann, 2022-01-31:
Unfortunately, I cannot reproduce this with AMD and Intel GPU drivers, and I don't have any Nvidia hardware.
What if you switch to the Nouveau driver?

Scumi, 2022-02-04:
> [5a1d779](https://github.com/XCSoar/XCSoar/commit/5a1d7797892d4729f916495924dc60a4681dbac0) introduces a different bug).

#858

MaxKellermann, 2022-02-04:
> Any idea how I could troubleshoot the Nvidia driver?

No, sorry. With proprietary software, I cannot help.

Scumi, 2022-02-04:
Okay. I found that compiling for rendering with OpenGL/ES 2.0 (GLES2=y) works.

MaxKellermann, 2022-02-04:
Hm, I wonder if we should default to GLES2 and remove desktop OpenGL support?

Scumi, 2022-02-05:
I still have to explicitly set GLES2 after 94c89f956de422056b7271fb9ff291e33c3e97e5 to not get a black window with my nvidia card.

SP-NTX, 2026-05-29:
Experiencing issue similar/the same to the one described, Debian 13, XCSoar downloaded from the website, black screen, then built myself also with GLES2 flag, yet still black screen, on NVIDIA Driver version 595

lordfolken, 2026-10-05:
Triage: the original “must set `GLES2=y`” path is obsolete — GLX desktop OpenGL was removed (#2140), and UNIX OpenGL is EGL + OpenGL ES by default.

Leaving this open for the remaining symptom: NVIDIA Linux black screen under EGL/GLES (see @SP-NTX’s May 2026 report with GLES2 still black on driver 595). Not closing solely because GLX is gone.

hor63, 2026-10-16:
Here from the output of the debug version on my NVIDIA card on Ubuntu 2404LTS with kernel 7.0.0-38-generic :

> GL vendor: Mesa
> GL version: OpenGL ES 3.1 Mesa 25.2.8-0ubuntu0.24.04.4
> GL renderer: **NVCF**
 
This is the open-souce NVIDIA driver according to my quick google search.

[image]

BTW: This screenshot was from the new GPU topography renderer from PR #3253. That works too.

hor63, 2026-10-16:
Are we talking about XCSoar on 

- X11
- Wayland
- bare metal DRI/KMS

Which GL driver (Mesa or custom NVidia)?

BTW: how much effort do we want to put into this issue? The issue is 4 years old on a kernel V5-something.
This platform is *not* intended to ever fly (desktop computer with an external monitor).
Useful of course for the armchair pilot in replay-mode, or trainings.

lordfolken, 2026-10-17:
Judging by the date, i'd say this was X11 on nvidia proprietary driver. 
Effort: Preferably none. This is a proprietary driver, that multiple other projects refuse to support. For example sway window manager. 
Both XCSoar, kernels, and the wayland support have moved considerably even since the last report in may. 
I'd say we close this for now. If there is someone that experiences this, they should also be prepared to debug and support their setup.

closed 2026-10-17 by lordfolken

Scumi, 2026-10-17:
I swapped my graphics card to non-NVidia a fair while ago, problem solved. I cannot say if the issue would still be there with newer drivers and Kernels. Agree to close.