XCSoar/XCSoar #850: XCSoar crashes at startup on Linux with SDL rendering

hm1702, 2022-01-29:
XCSoar versions having and not having the problem
-------------------------------------------------
Last working version https://github.com/XCSoar/XCSoar/commit/8565c62c3943d285b65ed74a26517e4e41d81ee1

Broken version https://github.com/XCSoar/XCSoar/commit/a8f8918f302d18b3685c1a61d5cc726422563aa2

Expected behavior
-----------------
Startup screen is showing

Actual behavior
---------------
XCSoar crashes immediately.
Debug build is showing
```
Starting XCSoar XCSoar v7.23-Linux~git#a8f8918f30
xcsoar: src/ui/window/Window.hpp:745: const PixelRect Window::GetClientRect() const: Zusicherung »IsDefined()« nicht erfüllt.
```

Steps to reproduce the behavior
-------------------------------
Build with
`make TARGET=UNIX OPENGL=n ENABLE_SDL=y DEBUG=y all -j 8`

Do you have any idea what may have caused this?
-----------------------------------------------
Wrong/missing definition of GetSize() in TopCanvas.hpp
SDL specific GetSize() is missing.

Do you have an idea how to solve the issue?
-------------------------------------------
Implement GetSize function for SDL rendering

MaxKellermann, 2022-01-29:
I can't reproduce this problem. Please bisect and post a backtrace.

> Wrong/missing definition of GetSize() in TopCanvas.hpp

If this were really missing, and somebody would use it, then XCSoar wouldn't even compile.

hm1702, 2022-01-29:
XCSoar compiles well and there is a GetSize().
It's defined in TopCanvas.hpp
```
#ifdef USE_MEMORY_CANVAS
  PixelSize GetSize() const noexcept {
    return PixelSize(buffer.width, buffer.height);
  }
#endif
```

`buffer` is defined here

```
#else /* !GREYSCALE */
  WritableImageBuffer<ActivePixelTraits> buffer;
#endif /* !GREYSCALE */
#endif /* USE_MEMORY_CANVAS */
```

The `buffer` variable is never initialized so GetSize returns nonsense
[image]

which finally results in a crash at varying places, depending on the contents of `buffer.`
Sometimes it even survives with a unreadable small font in the startup screen 

Here an example  traceback
[image]

MaxKellermann, 2022-01-31:
You are right - in certain configurations, the `buffer` field is never used, never initialised, but is used for `GetSize()`. This bug has existed forever, but was never a problem until we started using `GetSize()` in these configurations.

closed 2022-01-31 by MaxKellermann