XCSoar/XCSoar #855: The imported xci file to control the Bluefly vario has no effect anymore

PhilippeOppliger, 2022-01-31:
XCSoar versions having and not having the problem
Version 7.21 has this problem, 6.8 did not

Expected behavior
pressing the button in the new 'bluefly' menu should start an action

Actual behavior
nothing happens

Steps to reproduce the behavior
Load this file 'bluefly.xci' to XCSoar Data folder, if you uncomment the last block, XCSoar will crash on reboot

mode=Menu
type=key
event=Mode BlueFly
label=BlueFly
location=5

mode=BlueFly
type=key
event=SendNMEAPort1 BVU
label=Vol Up
location=1

mode=BlueFly
type=key
event=SendNMEAPort1 BVD
label=Vol Down
location=2

mode=BlueFly
type=key
event=SendNMEAPort1 RST
label=Reset
location=3

mode=BlueFly
type=key
event=SendNMEAPort1 BSD 1000 500 800 500 600 500 400 2000 1000 500 800 500 600 500 400 2000
label=TestBeep
location=4

mode=BlueFly
type=key
event=SendNMEAPort1 BFS 1000
label=Sink Sound Off
location=5

mode=BlueFly
type=key
event=SendNMEAPort1 BFS 100
event=SendNMEAPort1 BOS 20
label=Sink Sound On
location=6

mode=BlueFly
type=key
event=Mode default
label=Cancel
location=9

mode=default
type=gce
data=AIRSPACE_ENTER
event=SendNMEAPort1 BSD 1000 500 800 500 600 500 400 2000 1000 500 800 500 600 500 400 2000

#mode=default
#type=gce
#data=STARTUP_REAL
#event=SendNMEAPort1 BSD 392 100 784 100 392 100 784 100

 

Do you have any idea what may have caused this?
no

Do you have an idea how to solve the issue?
no

MaxKellermann, 2022-01-31:
Please try version 7.22, and try all versions between 6.8 and 7.21 to find out exactly which version regressed.

PhilippeOppliger, 2022-01-31:
it's running without problem with 7.22, Thanks a lot for the very quick help !!!!
(i run the xcsoar on a Kobo Clara HD, until now with Version 6.8.16 (was the best for Clara HD). I was very happy to see that you have a version supporting the Clara HD, looks very good, also WLAN is running and I can connect a Flarm unit over the WLAN. 
Thanks a lot

closed 2022-02-01 by MaxKellermann