XCSoar/XCSoar #857: Latest build fail to run on WIN 10 64 system

MikeBravo15, 2022-02-02:
XCSoar 7.22
-------------------------------------------------

Expected behavior
-----------------

Start the program

Actual behavior
---------------

XCSoar generate following error:
assertion failed!
File src/LocalPath.cpp
Line:80
Expression: !data_paths.empty()

Steps to reproduce the behavior
-------------------------------

Build XCSoar with  make TARGET=WIN64 and run under Windows 10 64

Do you have any idea what may have caused this?
-----------------------------------------------

No

Do you have an idea how to solve the issue?
-------------------------------------------

No

MaxKellermann, 2022-02-04:
Can you reproduce this with git master? I can't.

MikeBravo15, 2022-02-05:
I run latest Debian 11 on a WM virual machine. Today I made a clean pull from the github master, build and run UNIX version and build the a WIN64 version. When I copy xcsoar.exe to a flash disk and connect to the WIN 10, I can run .exe from the flash disk OK, but when I copy it anywhere elese, I got above mentioned error. Hope it helps

MaxKellermann, 2022-02-10:
If that happens again, c2e968e9b9bdadb2af3e7e5e2d54909c66ce3326 turns this into a runtime error instead of a crash.

closed 2022-02-10 by MaxKellermann