XCSoar/XCSoar #860: dlgCredits doesn't scroll in textfield

lordfolken, 2022-02-04:
XCSoar versions having and not having the problem
-------------------------------------------------
7.23

Expected behavior
-----------------

Be able to read the entire list of people that contributed to XCSoar in app. 
Be able to read the entire changelog
Be able to read the licence

Actual behavior
---------------
[image]

The text fields that are displayed are cut off once they are full, and accessing the rest of the text isn't possible.

Steps to reproduce the behavior
-------------------------------

info->info->credits->page2,3,4

Do you have any idea what may have caused this?
-----------------------------------------------

not implemented.

Do you have an idea how to solve the issue?
-------------------------------------------

Add scrollbar to text-field, when text doesn't fit into text-field.

Scumi, 2022-02-04:
On UNIX target scrolling works with arrow-keys, but only after clicking in the text area with the mouse.

lordfolken, 2022-02-04:
Thx. i didn't notice. I updated the title to be more precise.

Scumi, 2022-02-04:
Just wanted to provide my observations. On Android it doesn't work at all. 

I also re-read your issue and recognized that you probably mean horizontal overflow, while I was thinking of vertical overflow (which is also not accessible).

lordfolken, 2022-02-04:
Adding linebreaks according to geometry is another problem, you are right.

lordfolken, 2022-02-04:
The more i look at it, the more is wrong with it:
 - missing scrollbar for touch input
 - text isn't reformatted for the width
 - line-by-line scrolling instead of pagination (maybe 2/3)
 - Since this is the only element on this page, it shouldn't need to be necessary to focus it for scrolling up and down. 
 - Up / Down should paginate, left/right change the page. Esc close the dialog.

MaxKellermann, 2022-02-04:
The LargeTextWindow class is horribly incomplete, that's true. I wrote it 12 years ago for the Android port, and concentrated on other things when it barely worked just well enough to show something. It needs some love.
But the scrolling issue is fixed now.
For other improvements, there should be separate issue reports. (And of course a volunteer who wishes to implement them.)

closed 2022-02-04 by MaxKellermann

lordfolken, 2022-02-04:
Thx for the fix. I'll have a look at the rest.