XCSoar/XCSoar #868: Language selection menu is missing in UNIX build

MikeBravo15, 2022-02-07:
XCSoar versions having and not having the problem
-------------------------------------------------

7.23 UNIX build

Expected behavior
-----------------
SW GUI language can be chosen by user 

Actual behavior
---------------

In menu Look/Language, Input is missing menu Item Language, Not sure if it is a bug or per design choice for Linux
[image]

Steps to reproduce the behavior
-------------------------------

Open mentioned menu.

Do you have any idea what may have caused this?
-----------------------------------------------

No

Do you have an idea how to solve the issue?
-------------------------------------------

No

MaxKellermann, 2022-02-07:
That's a feature, not a bug. On Linux, XCSoar uses the standard language selection via LANG, LC_ALL etc.

August2111, 2022-02-11:
Please excuse my objection: but with embedded systems such as Cubie (OpenVario), Raspi etc., I think it is very bad that a language change only takes place in the environment. For example, is it not possible or difficult to store 2 different language settings in the profile settings on an airplane? Does this mean, for example, that a separate menu setting has to be created for each language in the OpenVario? Personally, I don't think this is a good solution and I ask that you reconsider!

MaxKellermann, 2022-02-11:
I don't understand. What do you want? What do you want to reconsider?
Technically, there's no difference between regular Linux and what you call "embedded". Both are just "Linux", and XCSoar implements the relevant Linux standard provided by the Linux system library.

August2111, 2022-02-11:
Of course you're technically right - but it's extremely unwieldy, e.g. in OpenVario, that I can't set different languages ​​for different pilots with the settings. And that leads to an extra list to be maintained in the OV menu!
I don't quite understand why XCSoar behaves differently on the UNIX systems than on all others

lordfolken, 2022-02-14:
why not keep it like on android? Automatic default, but possible to overrule.

August2111, 2022-02-15:
> why not keep it like on android? Automatic default, but possible to overrule.

This follows my suggestion exactly. About 2 weeks ago I had already prepared something in this direction - but without Max's consent it makes little sense to bring it up here.
 A decision would also have to be made as to whether this should be done in general for Linux/UNIX - or only on the embedded devices. But I would think a general solution for all targets would be best!

MaxKellermann, 2022-02-15:
> why not keep it like on android? Automatic default, but possible to overrule.

You can't combine both. Either you use libc's i18n code, or you use XCSoar's own i18n code.

XCSoar behaves like a good citizen and uses Linux's native language selection. On OpenVario, it even works without a reboot, after I fixed the menu code.

August2111, 2022-02-15:
I know what you mean. But that don't solve the problem of a multiuser usage, f.e. in a club: I cannot define my language in the profile! And I don't like to set it every time in the  (new) menu again and again. So I have to patching the code for us, but this I don't like really :(

closed 2022-03-02 by MaxKellermann