XCSoar/XCSoar #869: docker specs outdated

hjr, 2022-02-08:
I like the idea of build with the help of docker. Specifically ANDROID and KOBO cross compilation targets.
Unfortunately I am not able to run the docker build at all any more.

UNIX
---------------
docker run     --mount type=bind,source="$(pwd)",target=/opt/xcsoar     -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest xcsoar-compile UNIX
  CXX     output/UNIX/dbg/src/Device/Driver.o
g++: error: unrecognized command line option '-std=c++20'; did you mean '-std=c++2a'?
g++: error: unrecognized command line option '-fcoroutines'; did you mean '-fc-prototypes'?
make: *** [build/compile.mk:108: output/UNIX/dbg/src/Device/Driver.o] Error 1
make: *** Waiting for unfinished jobs....
rm output/UNIX/dbg/src/Device/Driver/../dirstamp
>> after changing the c++20 to c++2a:
docker run     --mount type=bind,source="$(pwd)",target=/opt/xcsoar     -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest xcsoar-compile UNIX
  CXX     output/UNIX/dbg/src/Device/Driver.o
g++: error: unrecognized command line option '-fcoroutines'; did you mean '-fc-prototypes'?
make: *** [build/compile.mk:108: output/UNIX/dbg/src/Device/Driver.o] Error 1
make: *** Waiting for unfinished jobs....
rm output/UNIX/dbg/src/Device/Driver/../dirstamp

KOBO
---------------
>> after installing meson by hand ..
root@85380812e7b6:/opt/xcsoar# xcsoar-compile KOBO   
make: Nothing to be done for 'libs'.
Package alsa was not found in the pkg-config search path.
Perhaps you should add the directory containing `alsa.pc'
to the PKG_CONFIG_PATH environment variable
No package 'alsa' found

ANDROID
---------------
root@85380812e7b6:/opt/xcsoar# xcsoar-compile ANDROID
make: /root/opt/android-ndk-r23b/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++: Command not found
>> I do see a /root/opt/android-ndk-r22 directory

ALL
-------------------------------
python complains about the removesuffix() not available, even on 3.10., so the interpreter should have this str method. Replaced it by the long term known rstrip() and it worked.

Steps to reproduce the behavior
-------------------------------
1. XCSoar from repo head.
2. Run the docker file generation script.
3. Run the build commands for targets.

Do you have any idea what may have caused this?
-----------------------------------------------
Some clues as mentioned above.

Do you have an idea how to solve the issue?
-------------------------------------------
Have 5 min. of a docker and tool xcs chain specialist .. I am not the one unfortunatley.

lordfolken, 2022-02-08:
are you sure that what you are using is current? The pipelines and so are all running in docker, and not one of the targets fail on the current commit.

lordfolken, 2022-02-08:
https://github.com/XCSoar/XCSoar/actions/runs/1812632908/workflow for example.

lordfolken, 2022-02-08:
so i rebuilt the container, and built android kobo and unix targets. I'm sorry i can't reproduce. 

All the issues you are describing, are pointing to an old image. Potentially do a docker images and see which is the newest one.

hjr, 2022-02-09:
Ok, will redo the trial. Is there any requirement on the host system?

hjr, 2022-02-09:
Another simple solution to get recent build would be access to the CI pipelines on github. Have not found any link to that. Is there any?

hjr, 2022-02-09:
Thx a lot. 
Deleted all docker images and containers and then again did the docker image build. After 3h of crunching and downloading it all works.

nice :)

closed 2022-02-09 by hjr